betaflight/betaflight #4091: OSD Warn if in turtle mode

spuder, 2017-09-08:
Show in the OSD a big warning if anti-turtle mode is engaged. 

I watched someone nearly loose their new quad off a cliff because a misconfigured aux channel had the quad stuck in anti-turtle mode. It took a long time to troubleshoot. 

It would be nice to have additional feedback/warnings when anti-turtle mode is engaged.

DanNixon, 2017-09-08:
I'll take a look this weekend.

etracer65, 2017-09-08:
Maybe also add some kind of unique beeper/status LED pattern for those that don't have a BF OSD.

Arakon, 2017-09-08:
Agreed, since it may also warn people to not try and turn it over if they see it flipped over in a park, for example. Maybe a slow "beeeep ------ beeeeep" kind of pattern.

spuder, 2017-09-08:
bonus points if it beeps the teenage mutant ninja turtle theme song

DanNixon, 2017-09-08:
Beeping does not indicate "do not pick me up" at all.
I leave my beeper on when I'm walking to my quad and quite a few times others have picked it up for me (obviously just trying to be helpful, but I'd much rather they just left it be).

That being said I'm not against adding a beep pattern for it.

isilcala, 2017-09-26:
For now, I set beeper & turtle mode on the same switch so it beeps when in turtle mode. But yeah an OSD warning is needed.

PR #4112 (2017-10-11): Anti turtle mode warnings
Adds OSD warning and beeper pattern any time anti turtle mode is active.

Feel free to suggest a better warning text for the OSD.

Fixes #4091
closed 2017-10-11 by mikeller