betaflight/betaflight #4094: BF 3.2 setpointweight is disabled by default #4017

jirif, 2017-09-08:
This is issue #4017 reopened

@borisbstyle
> 100 is actually not doing nothing (disabled) so it means it has no point to do anything
> relaxFactor = 1.0f / (pidProfile->setpointRelaxRatio / 100.0f);

issue #4017 is propably too complex so dont hurry to close it again, i spent great time to figure out why 3.2 flying like bs with setpointRelaxRatio 1. The problem is not with setpointRelaxRatio but dtermSetpointWeight so lets evaluate the code step by step:
```
float dynC = 0;
if ( (pidProfile->setpointRelaxRatio < 100) && (!flightModeFlags) ) {
    dynC = dtermSetpointWeight * MIN(getRcDeflectionAbs(axis) * relaxFactor, 1.0f);
}
const float rD = dynC * currentPidSetpoint - gyroRateFiltered; // cr - y
```
When pidProfile->setpointRelaxRatio == 100 whis is default value dynC stay at zero and rateerror is calculated incorrectly because currentPidSetpoint is multiplied by zero so
rD = **0 - gyroRateFiltered**
dtermSetpointWeight is ignored and this turn 2dof pid controller off ... lol
dynC must be set at leas to value of dtermSetpointWeight also when relax ratio is 100  (with default values dtermSetpointWeight = (60/127) )
**dynC = (60/127) * 1**
then rate error will be calculated correctly
**rD = (60/127) * currentPidSetpoint - gyroRateFiltered**

This way it worked before your commit https://github.com/betaflight/betaflight/commit/45c578c8b65ac064aaf0ccdf833c49cc1c506479 and not working after because of incorrect optimalization of this code:
```
float dynC = dtermSetpointWeight;
if (pidProfile->setpointRelaxRatio < 100) {
    dynC *= MIN(getRcDeflectionAbs(axis) * relaxFactor, 1.0f);
}
const float rD = dynC * currentPidSetpoint - gyroRateFiltered;    // cr - y
```
Pull request #4076 is optimal solution ...

martinbudden, 2017-09-08:
@jirif , is not correct solution:
```
float dynC = dtermSetpointWeight;
if ( (pidProfile->setpointRelaxRatio < 100) && (!flightModeFlags) ) {
     dynC *= MIN(getRcDeflectionAbs(axis) * relaxFactor, 1.0f);
}
const float rD = dynC * currentPidSetpoint - gyroRateFiltered;    // cr - y
```

borisbstyle, 2017-09-08:
@jirif 
yes with 100 you basically have error from gyro ;). (0 - gyro)

0 = fully enabled
100 = disabled

borisbstyle, 2017-09-08:
Oh wait. So some optimisation inverted the meaning of the code you are saying?

borisbstyle, 2017-09-08:
so this should indeed be the correct code

```float dynC = dtermSetpointWeight;
if (pidProfile->setpointRelaxRatio < 100) {
    dynC *= MIN(getRcDeflectionAbs(axis) * relaxFactor, 1.0f);
}
const float rD = dynC * currentPidSetpoint - gyroRateFiltered;    // cr - y```

borisbstyle, 2017-09-08:
This means that by default people were flying in rate from gyro mode as relaxratio was 100.

Therefore we need to readjust the defaults.

borisbstyle, 2017-09-08:
I am not sure what to do with it as it will change default flight behaviour what people are used to in the current versions of release candidates

jirif, 2017-09-08:
@martinbudden 
this will revert https://github.com/betaflight/betaflight/commit/45c578c8b65ac064aaf0ccdf833c49cc1c506479 because when a flightmode flag is set (non ACRO modes) we need disable 2-dof so dynC must be zero

@borisbstyle 
yes it changed code meaning

as i said for martinbudden this will revert your fix from https://github.com/betaflight/betaflight/commit/45c578c8b65ac064aaf0ccdf833c49cc1c506479 
if you like to have conditions the code should look like this but i think it will be suboptimal

```
float dynC = 0;
if (!flightModeFlags) {
    dynC = dtermSetpointWeight;
    if (pidProfile->setpointRelaxRatio < 100) {
        dynC *= MIN(getRcDeflectionAbs(axis) * relaxFactor, 1.0f);
    }
}
const float rD = dynC * currentPidSetpoint - gyroRateFiltered;    // cr - y
```
i still think think my original solution look better :)
```
float dynC = 0;
if ( (pidProfile->dtermSetpointWeight > 0) && (!flightModeFlags) ) {
    dynC = dtermSetpointWeight * MIN(getRcDeflectionAbs(axis) * relaxFactor, 1.0f);
}
const float rD = dynC * currentPidSetpoint - gyroRateFiltered;    // cr - y
```

borisbstyle, 2017-09-08:
See this one here. I also corrected the default parameters and bumped the eeprom so the import of configuration generated with "diff" command would produce same flight behaviour
https://github.com/betaflight/betaflight/pull/4095

borisbstyle, 2017-09-08:
i used your code, just changed defaults and eeprom version

jirif, 2017-09-08:
@borisbstyle 
dynC must be zero when a flightmode is set, have look at my previous post

borisbstyle, 2017-09-08:
it is ;)

martinbudden, 2017-09-08:
PR looks good. Let's merge it in so it is available for RC5.

jirif, 2017-09-08:
@borisbstyle
yes it is by default but we need force it to be zero when a level mode is on.

borisbstyle, 2017-09-08:
@jirif Your code is exactly as mine was ;)
I even merged your PR and mine on top of it so the commit goes to you

closed 2017-09-08 by borisbstyle