betaflight/betaflight #4104: Feature request: Vtx switch (adc switch)

jacobbloy, 2017-09-09:
Similar to camera settings Being changed via taranis switches. 
Can an aux channel be used to change vtx channel. I know smart audio is there but not on all vtx, the buttons are annoying and by soldering the one wire to an adc port then grounding it, you could map an AUD switch to change your vtx channel or power.

jflyper, 2017-09-09:
@jacobbloy 
I didn't quite understand what ADC has to do with VTX control, but using AUX switches and stick commands to control band/channel/power for supported intelligent VTXs (SmartAudio, Tramp and OpenTCO --- planned) is coming, probably in 3.3, if this is what you want.

You can search in issues with a keyword "vtx" for collection of features that others are requesting. Some will be implemented in v3.3, and others are still under discussion.

jflyper, 2017-09-09:
@jacobbloy 
BTW, there is a limited support for stick commands and AUX switches.

See
http://singularity.impulserc.com/assets/files/manual.pdf
for details. Stick commands and AUX channel control might work for you.

closed 2017-09-09 by jflyper

DanNixon, 2017-09-09:
I have a feeling this is about VTXes with a single button to change modes (still nothing to do with ADC though).

There are several issues with supporting these and no good reason to purchase a VTX that does not support either SmartAudio or IRC Tramp protocols.

jflyper, 2017-09-09:
Ah, okay, the old story of "Substituting a vtx button with GPIO".

DieHertz, 2017-09-09:
The VTX with a push button are good candidates, I want to play with this on my ft48x

jflyper, 2017-09-09:
@DieHertz How are you planning to going to get feedbacks?

DieHertz, 2017-09-09:
I'm not going to, camera control doesn't have feedback either)

jflyper, 2017-09-09:
Oh, the cam control has visual feedback;)

DanNixon, 2017-09-09:
You have the camera OSD for feedback.

With these VTXes the only feedback is either an indicator on the device or adjustment until you hit the right channel.

It would be an interesting use of a generic GPIO feature, but I doubt they could be integrated with the current VTX specific functionality.

DieHertz, 2017-09-09:
No sure not, I just wanted it to have ability to e.g. c cycle power and channels. My push button is not accessible )

mikeller, 2017-09-10:
Yeah. While not being super comfy, being able to use a RX channel to simulate button pushes will be a lifesaver on builds where the button cannot be reached.

jacobbloy, 2017-09-10:
i have many builds that don't have smart audio ect, and many micro builds that use an AIO vtx, it is not always possible to use a smart vtx. this is why i have put in the request. I'm looking into a solution.

jacobbloy, 2017-09-10:
I have adjusted the beep sequence for my AUX beeper. To 20ms and no hold time this creates a long beep that stops after about 20ms from releasing the switch. This allows me to do a short vtx button press and a long press. The timer isn’t dependent on the switch and is constant so the time it takes for the beep to stop isn’t exactly 20ms. Every few cycles it has a break in the long press that could be detected as a double press of the bottom so it will need to be cleaned up.

My recommendation is to add vtx_button =ON/OFF in cli that changes the beeper AUX beep cycle so as I have done it can emulate a vtx button press. As simple to setup as one wire from beeper pad on one pad on vtx button. In most cases you wouldn’t even need to remove the vtx button.

My coding is limited to changing values and building the firmware. But I will playing and testing.

jacobbloy, 2017-09-10:
At this poin no visual of your vtx channel and power can be displayed or detected. In the future you could have a vtx channel base point and power base point and the ability to set button press sequence to change channel/band/power. Sad thing is all vtx are different. Personally I do not want this feature to be a fully intergrated system just for convineance of not pulling apart my quad.