betaflight/betaflight #4107: BF 3.2RC4 - beeper -ON_USB prevents gyro calibration beeps

etracer65, 2017-09-09:
With a clean install of BF 3.2RC4 and all settings at defaults, switching off beeps when on the USB connection (beeper -ON_USB) will prevent the gyro calibration completion beeper sequence even when powered from battery (USB disconnected).  Re-enabling the beeper on USB will restore the gyro beeps when only on battery.  This behavior changed since 3.1.7.

I noticed this because I normally connect the battery on my bottom-mounted quads and then set them down to allow the calibration to finish.  I'm tuned into the 3-beep sequence.

So hopefully this is just a beeper issue and not an actual gyro calibration issue.  I normally use the `set gyro_cal_on_first_arm = ON` option as well so I haven't had any problems.

In my case the target is AIRBOTF4.

etracer65, 2017-09-09:
Update the title because I found that the gyro calibration beeps will play intermittently on battery connect.  Not consistent and I'm not able to discern a pattern.  Happening with multiple quads all running BF 3.2RC4 with the AIRBOTF4 target.  Enabling the beeper on USB is a workaround, but then it's very unpleasant when connected to the configurator it the radio is off. :)

etracer65, 2017-09-10:
Additional report: https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread/page3442#post38254061

mikeller, 2017-09-10:
Had a quick look into this. There is a bug in the firmware that prevents the beeps when `gyro_cal_on_first_arm` is on. This is fixed in #4109.

The cause for the calibration beeps on startup not working (intermittently) is that the battery presence detection has been improved, and the limits for when the battery voltage is recognised as stable and cell count detection is done have been tightened. This means that after connecting the battery, the gyro calibration now is often finished before the battery cell count detection is finished. Since an USB connection is only detected by the voltage on the board being outside of the range for LiPo batteries, this means that the gyro calibration beeps are suppressed since the system is in a 'no battery detected, must be on USB' state at the time.

You can verify this yourself by holding your craft in your hand when connecting the battery, and then putting it down. The delay in gyro calibration introduced by the movement of the craft being above `moron_threshold` will be enough to let the battery detection finish before the gyro calibration, and the calibration beeps will work.

This is a bit of a drawback, but I am not sure if it merits the added complexity of queueing beeps while the battery calibration is running.

etracer65, 2017-09-10:
@mikeller  I tested with PR4109 and that fixed the `gyro_cal_on_first_arm` beeps.

Regarding the initial gyro calibration beeps: perhaps a solution would be to have the gyro calibration beeps ignore the ON_USB setting.  Having the 3 beeps even when connecting to USB wouldn't be obtrusive.  Many people (myself included) key into those gyro calibration beeps as an indicator the flight controller is "ready to go" so it causes confusion when they don't happen.  I would consider the initial gyro calibration beeps part of the initialization of the flight controller.  The precedent for having them override the ON_USB setting is the SYSTEM_INIT beeps as they already override the ON_USB settings and always play.  If somebody **really** doesn't want the gyro calibration beeps they can always disable them with `beeper -GYRO_CALIBRATED`.

etracer65, 2017-09-10:
Or perhaps go even further and only override the ON_USB setting if the battery cell detection is still underway.  But this would possibly cause an intermittent effect in reverse where sometimes the gyro beeps happen on USB when they're not suppose to because of -ON_USB.

etracer65, 2017-09-10:
Created https://github.com/betaflight/betaflight/pull/4115 to allow GYRO_CALIBRATED beeps to override the -ON_USB setting.  Tested on my quads and works as expected.  The initial gyro calibration beeps are consistent and behave as in earlier firmwares.

jirif, 2017-09-10:
I posted another approach to fix this problem #4121

mikeller, 2017-09-10:
I thinkd #4121 is a good solution for this, thanks @jirif.