betaflight/betaflight #4111: GPS

djpret, 2017-09-10:
I am trying to setup OSD with GPS information using a NAZE32 Rev6 Acro FC and a microminim OSD.  I was using Betaflight firmware 3.1.7 and now have tried the new (preview) 3.2 version.  

I don't want the GPS to assist with control of the quadcopter, but simply to provide GPS info (speed, direction and distance to home, etc) to the OSD.  The OSD is setup through a soft serial and is working, however I cannot get any GPS info.  The GPS is wired into the UART #2 and powered from 5V.  All connections are good and have been triple checked.

I am somewhat new at scratch building quads and have looked this issue up a lot, but still am not 100% sure that I have not missed something.  

My guess is that my Flight controller does not support a GPS input.  Can anyone confirm or deny this, and is there a way to get GPS info onto my OSD with a NAZE32 board and microminim OSD while using BetaFlight firmware?

Thanks,
DaveP

Arakon, 2017-09-10:
I'm fairly sure GPS was disabled for F1 boards since they are out of space and processing power. You can probably disable some features and enable GPS again if you compile it yourself.

jflyper, 2017-09-10:
@djpret 
GPS feature inclusion by default is for targets with **_more_** than 256K flash (usually F4 and F7).
https://github.com/betaflight/betaflight/blob/b2cd7294e9d392cf96c360ffac5fc51e5b93ba3d/src/main/target/common_fc_pre.h#L135-L137
```
135 #if (FLASH_SIZE > 256)
136 // Temporarily moved GPS here because of overflowing flash size on F3
137 #define GPS
```
On NAZE, with STM32F103CB that only have 128K+ flash, GPS is not included.

As a side note, please bare in mind that F1 support is coming to EOL.

closed 2017-09-10 by jflyper