betaflight/betaflight #4119: F4 Overclock Can't Bind Spektrum

brucesdad13, 2017-09-10:
1.) Recently I've been assembling a batch of about 50 flight controllers some F4 and some F7. The F4s I decided to overclock by default. I had issues binding spektrum and assumed it was a hidden solder bridge or bad tweezers... I went all around the block assuming hardware or user error. Then the idea occured that perhaps it had to do with overclock. If others could test on their F4 with Spektrum / compatible (e.g. Lemon) that would be most appreciated.
2.) Steps to reproduce the problem
  - Flash 3.2 latest master and attach DSM2/DSMX sat.. jumper bind pin
  - set serialrx_provider = SPEK2048 or SPEK1024 or SRXL
  - set spektrum_sat_bind = (whatever is appropriate for your rx)
  - set cpu_overclock = ON
  - save
  - Power on fc with bind button depressed or pads jumpered and in my case the satellite doesn't enter bind mode
  - set cpu_overclock = OFF
  - save
  - Power on fc again with bind button depressed or pads jumpered and satellite enters bind mode correctly

3.) ALIENWHOOPF4 flight controller with various Spektrum and Lemon RX DSM2 and DMSX satellites soldered UART3 etc as per target.h
4.) Firmware compiled from latest master (3.2) and configuration stock for ALIENWHOOPF4 target except cpu_overclock = ON
5.) diff:
```
# diff

# version
# Betaflight / ALIENWHOOPF4 (AWF4) 3.2.0 Sep 10 2017 / 14:12:45 (2b72d93)

# name

# resources

# mixer

# servo

# servo mix

# feature

# beeper

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master
set cpu_overclock = ON

# profile
profile 0

# rateprofile
rateprofile 0
```

brucesdad13, 2017-09-10:
The workaround is to bind before overclocking.

jflyper, 2017-09-10:
@brucesdad13 Can you scope the bind signal?

brucesdad13, 2017-09-11:
The scope shows no pulses issued to start the bind process when set cpu_overclock = ON. According to Spektrum docs, "To put a receiver into bind mode, within 200ms of power application the host device needs to issue a series of falling pulses. The number of pulses issued selects which bind types will be accepted from the transmitter. " Here's the LA screenshot with overclocking (no pulses to trigger bind mode):

[image]

Without overclocking:

[image]

Closeup of 9 pulses for spektrum_sat_bind = 9:

[image]

I believe I have the correct settings "The packet is transmitted at 125000bps, 8 bits, No parity, 1 stop (8N1)." 

[image]

jflyper, 2017-09-11:
I suspect this is caused by malfunctioning "power cycle detector"; there is a logic that checks if power cycle has occurred to make sure the receiver has power cycled, too.

mikeller, 2017-09-11:
Sounds quite right, @jflyper. Overclocking is implemented by setting the clock registers, and then doing a soft reset. The code for Spektrum bind will probably have to be amended to take this into account.

jflyper, 2017-09-11:
I think we a `isMPUSoftReset` that handles OC case... but I have no idea at the point 😭

jflyper, 2017-09-11:
Something like `BABEFACE`!?

brucesdad13, 2017-09-11:
I've always been a fan of DEADBEEF

brucesdad13, 2017-09-11:
OCF4C0DE

AndersHoglund, 2017-09-11:
Don't use the SerialPort analyser for this. It is just a number of pulses at startup, number depending on your `spektrum_sat_bind` setting.

brucesdad13, 2017-09-11:
Thank you @AndersHoglund ! @jflyper had also explained this to me. Either way, overclock is resulting in bind pulses not being sent. Can we move the bind code ahead of the overclock in fc_init?

AndersHoglund, 2017-09-11:
Don't think that moving the overclock code would help, as it does a `NVIC_SystemReset()`. The soft reset flag does probably not survive that. ( `RCC_CSR_SFTRSTF` in `isMPUSoftReset()`)

brucesdad13, 2017-09-11:
Can we skip the overclock reset if bind action called for and assume they'll power cycle the FC after bind before flying anyways?

PR #4132 (2017-09-13): Fixed Spektrum binding when overclocked.
Also fixes button detection when overclocked. (Both of these things need to happen before the soft reset to enable / disable overclocking, since they are triggered by `!isMPUSoftReset()`.)
Also fixed input format for `spektrum_sat_bind_autoreset`.

Fixes #4119.
closed 2017-09-13 by borisbstyle