betaflight/betaflight #4122: BF 3.2RC4 Serial Rx prevents bootloader mode (DFU) from working

etracer65, 2017-09-10:
While investigating inconsistent behavior around entering DFU mode while flashing firmware versions, I discovered that if the flight controller is receiving serial rx data from a bound receiver it will not enter DFU mode.  The flight controller must be powered cycled to have the normal VCP MSP connection return.  This occurs with both the firmware flasher and the CLI "bl" command.   If the receiver is connected and powered but the TX is turned off, then the bootloader can be entered normally.

Target: AIRBOTF4
Serial Rx: UART1 - SBUS (FrSky X4RSB)

mikeller, 2017-09-10:
This is a known 'feature' of the STM32 family: When booted into DFU / boot loader mode, they will listen on a number of ports (VCP / UART, model dependent), and assume that the first port they receive something on is the port used for communication while in DFU / boot loader mode. See this for details: 

http://www.st.com/content/ccc/resource/technical/document/application_note/b9/9b/16/3a/12/1e/40/0c/CD00167594.pdf/files/CD00167594.pdf/jcr:content/translations/en.CD00167594.pdf

This means that an RX sending a data stream on UART1 (used for DFU by most MCUs) will cause the MCU to bind to UART1 when entering DFU / boot loader mode.

etracer65, 2017-09-10:
Ok, makes sense.  I might suggest adding something about this to the configurator firmware flashing tab.  Possibly something about ensuring that the receiver is disconnected or powered off.  If not possible (because rx is direct connected and powered by flight controller) ensure that transmitter is powered off.  Might minimize some confusion for new users wondering why they can't update the firmware.

mikeller, 2017-09-10:
Something like this:

[image]

You're the second person in 2 weeks asking for the existing warning to be added. Proof positive that users don't read warnings. :-P

etracer65, 2017-09-10:
You're right - I missed that.  But I did actually read the page but I focused on the "Recovery/Lost communication" section because that's the logical place to see why it's not working.  Also with the trend to direct wire everything now it's not really possible or practical to disconnect the receiver.  So the additional information of having the TX powered off might save some headaches.

mikeller, 2017-09-10:
Yeah, there is serious information overload going on on that tab.

I think the thing that is really needed is to make users more aware of these considerations **at build time**. Not connecting serialRX to UART1 if this can be avoided will keep these problems from occurring in the first place...

etracer65, 2017-09-11:
Well that would be on the board designers as the builders generally don't have any option but to use the connection with the hardware inverter (at least on the F4).

jflyper, 2017-09-11:
@etracer65 
A thing that make this problem more complicated is that some receivers are not _**chatty**_ until they are first linked to transmitters after power cycle.

What people experience is that the FC suddenly becoming unresponsive to DFU command/button when they try flashing with transmitters turned on. They usually do not power off transmitters while desperately repeating FC power cycle and DFU attempts.

Oh, and hardware inverters on F4 seems not capable of stopping this woes. The boot loader seems to pick up any input that UART has detected, even a framing error event.

AndersHoglund, 2017-09-11:
I have seen issues even with Tx off, with auto bind Rx. Not sure how, can not repeat that now..... 
The only solid solution would be HW gated signal or power, under SW control. Not sure how solid that would be, as the bootloader knows nothing about it. Would depend on a default pin state at reset (hard and soft).

rodge402, 2017-12-21:
I'm glad I found this post because I flashed my F4 after having issues getting into DFU mode. A friend was binding the TX and RX at the time so it did flash and everything seemed ok until I noticed the channel mapping was acting up and not right. At first yaw was being controlled by a switch and out of no where it started working on stick movement. But the channels were still not right. I seen that in modes tab there were no aux channels. It just showed channel 1,2,3,4,5,6,7,8 and so on instead of channel 1,2,3,4,5,6 aux1, aux2, aux3 and so on like normal. Flashing at the same time as binding must have caused a glitch in the firmware. I plan on flashing the board again tomorrow and I'm sure it will flash correctly this time now that I know the RX can't be on at the same time. I can be wrong but I'm hopeful this saved me from giving up on this fc.

closed 2017-12-23 by mikeller

rodge402, 2017-12-23:
Thanks for closing this now that it could have help someone else. Lol

mikeller, 2017-12-23:
@rodge402: It **will** help others - issues stay around after they are closed. But it is not an actual issue (i.e. something that needs to be fixed), so the status should be closed.