betaflight/betaflight #4128: Playing with yaw spins, quad throttles down and falls out of sky ?

robert-hoffmann, 2017-09-11:
Don't really know what to think ..this has happened 3 or 4 times now.

The only things i have touched since this started happening is setting anti-gravity to 1.5 (BFC) and upping pid loop to 8k.

Also it happened only with crappy beat up props. The BBlog at the end, i was basicly just hovering and playing with yaw, this happened 2 times in a row. When i put clean props on it didn't happen a third time. And yesterday (no bb log unfortunately) it happened when i was doing a roll dive ...that didn't go so well ! :-)

It felt like the throttle was reduced progressively (like withing 2secs) until it just was on the ground ..except i didnt cut the throttle ^^

...any clues anyone ?

Thanks !

`# version
# Betaflight / MATEKF405 (MKF4) 3.2.0 Aug  6 2017 / 13:30:42 (981df1d54)

# name
name Chameleon

# resources
resource BEEPER 1 C13
resource MOTOR 1 C06
resource MOTOR 2 C07
resource MOTOR 3 C08
resource MOTOR 4 C09
resource MOTOR 5 NONE
resource MOTOR 6 A08
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 A15
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource LED 1 B09
resource LED 2 A14
[... 535 lines cut ...]
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83

# rateprofile
rateprofile 0

set rc_rate = 90
set rc_rate_yaw = 60
set rc_expo = 0
set rc_expo_yaw = 0
set thr_mid = 35
set thr_expo = 50
set roll_srate = 83
set pitch_srate = 83
set yaw_srate = 82
set tpa_rate = 0
set tpa_breakpoint = 1650`

[LOG00002.zip](https://github.com/betaflight/betaflight/files/1293776/LOG00002.zip) version

mikeller, 2017-09-11:
This almost sounds like low battery cutoff (#3510) kicked in. @brycedjohnson any idea if this could happen with this configuration?

brycedjohnson, 2017-09-12:
I don't see it on the diff so that should mean it should have been disabled (battery cutoff stays at 100%).

The CLI param is vbat_cutoff_percent.  If it is 100 it is disabled.

robert-hoffmann, 2017-09-12:
Just noticed in dump `set acc_trim_pitch = 2`

I don't remember setting that ...what is it ? I set it back to 0 for now...

mikeller, 2017-09-12:
These are the accelerometer trims (see https://shipow.github.io/cleanflight-web/assets/images/StickPositions.png). You probably 'trimmed' accidentally by having your sticks in an odd position.

robert-hoffmann, 2017-09-12:
> his almost sounds like low battery cutoff (#3510) kicked in

Batteries where all fully charged in any case.

Did a bunch of batteries today, and all was fine ..will try again with my crappy props later this week, see if i can reproduce with just that