betaflight/betaflight #4129: F7: Dshot output is broken

jflyper, 2017-09-12:
#3978 seems to have broken Dshot output.

Omnibus F7: No spin at all. With some rates, ESCs don't even initialize. Tiny twitch upon FC reset.
NERO: Spins twichingly until half throttle, grainy spin on entire range.

This whole section added by #3978 is causing this issue.
https://github.com/betaflight/betaflight/blob/master/src/main/drivers/pwm_output_dshot_hal.c#L86-L89
```
 86     if (motor->timerHardware->output & TIMER_OUTPUT_N_CHANNEL) {
 87         HAL_TIMEx_PWMN_Stop_DMA(&motor->TimHandle,motor->timerHardware->channel);
 88     } else {
 89         HAL_TIM_PWM_Stop_DMA(&motor->TimHandle,motor->timerHardware->channel);
 90     }
```

brianlbalogh, 2017-09-12:
Is this just related to the use of the timer N-channels? I haven't actually flown any builds since that commit yet, but my F7's seem fine on the bench with a current build.

jflyper, 2017-09-14:
@brianlbalogh No, it's not related with N-channel. (And I found out that NERO does not have a problem also).

goodnerdboy, 2017-09-14:
@jflyper my NeroF7 has a problem too. I don't know what is your setup with it, but i use DSHOT1200, and motor 1 is very grainy on most of throttle range(i didn't go full throttle though), motor 3 is twitching  without spinning, and motors 2 and 4 don't spin at all.
It seems to be related. I installed build 1619 (right before merging #3978 ) and it works well.
My 2 cents.

jflyper, 2017-09-14:
Hmm... I re-flashed NERO with RC5, and it develops the issue again. I wonder what's happening...

goodnerdboy, 2017-09-14:
does this look like a conflict issue? Did you modify any parameters before it works the first time?

wesk-ms, 2017-09-15:
Hey guys,

FWIW, I've had a omnibus f7 v2 from RTFQ now, for 2 weeks and it hasn't  been in the air yet because the motors won't spin.  What's interesting is that have been able to get all four motors working on any dshot protocol in the BF configuator motors tab if I drop the graph refresh date down to 1 sec.  If I do this, then disable/re-enable the motors, one motor will start working.  Disable/re-enable the motors a few more times, and I'll pick up another motor.  Doing this over and over for 3-4 minutes results in all four motors working.  Now, they aren't running well at all, but they're all spinning.  Very grainy and jerky at low speed.  At midrange, they all sound like they have bad bearings (motors have never been flown).  Disconnect the battery and I get a strange clicking sound from the FC, not from the motors and the beeper is disabled.  Disconnect the USB, then power back up and I'm back to no motors spinning and can go through this scenario again.  100% reproducible for me.

Anyway, thought it was interesting that changing the refresh rate on the graph would have this result.

I haven't looked into the BF code at all, nor have I ever cracked open the F7 processor manual, but this behavior reminds me of some critical section blocking I've seen in the Intel world.

My ESC is the 4in1 Typhoon blheli32 from RTFQ.

Happy to provide data, videos, test, or help in any way I can.

Thanks!!

This behavior is 100%

bikenewby, 2017-09-16:
Same problem on OMNIBUSF7 with DSHOT setting, the ESC won't initialize on RC5 and hence cannot arm. It is working fine if rolled back to RC4. Also, it is working with RC5 but with https://github.com/betaflight/betaflight/blob/master/src/main/drivers/pwm_output_dshot_hal.c#L86-L89 reverted to before #3978.

sdotfpv, 2017-09-18:
works on RC4 for the omnibus f7 thats what i switched to

jflyper, 2017-09-18:
Fix is coming. #4165

jamborloi, 2017-09-19:
@jflyper any news for the fix for the dshot? Currently building with omnibus F7 with wraith32 35a ESC it will be good to know if Betaflight #200 has the fix already.

jflyper, 2017-09-19:
#4165 is under review. You can check if it has been merged by clicking the PR link and check if it says "Merged", rather than "Open".

goodnerdboy, 2017-10-01:
Hi @jflyper.
Reading #4165  comments, i don't find it clear weither or not this will be fixed before BF3.3.
Do you think there is a chance we'll have a fix before 3.3?

SkylionFPV, 2017-10-02:
is this fixed in rc6 or not?

goodnerdboy, 2017-10-02:
@SkylionFPV it is not fixed yet in RC6, at least for NeroF7.

SkylionFPV, 2017-10-02:
thank you I was about to flash my omnibus f7s over but I figured I should ask first

SkylionFPV, 2017-10-08:
is this problem fixed now?

goodnerdboy, 2017-10-08:
It isn't fixed yet.
#4165 was supposed to fix it but wasn't merged so F7s are stuck without Dshot for the moment, or use older RCs (RC4 was the last one with Dshot working)

jflyper, 2017-10-21:
Closed via #4165

closed 2017-10-21 by jflyper