betaflight/betaflight #4130: Qx 105 Frsky no stick movements in Betaflight or cleanflight

jkapdrone, 2017-09-12:
I dont have any stick movements in the receiver section. The qx105 is new haven't flashed or upgraded anything. Working with frsky and my taranis is all setup. I followed everything in the manual. Please help with some ideas.

jkapdrone, 2017-09-12:
https://youtu.be/C_rQUbwjXCc

wind0r, 2017-09-12:
As written in the Issue Template this issue tracker is for development purpose. 
This is a Configuration Issue. Please ask the manufacture for help or in any forum.
Since we all develope betaflight in our spare time manufactures should atleast give support to the customers. They are selling stuff and we doing support and firmware code is just stupid.

Video shows the current Cleanflight Configurator which is not compatible with the Firmware Version you use. You should use a compatible configurator or upgrade the firmware.

closed 2017-09-12 by jflyper

jkapdrone, 2017-09-12:
My apologies....thank you