betaflight/betaflight #4136: Camera control Foxeer monster v2 - only second osd menu available.

csbuzkp, 2017-09-12:
I have camera foxeer monster v2, Matek f405 osd FC, connected osd pin to motor 6. Did configs - map resource motor 6 to camera control, change internal resistance to 99. When I hold throttle stick low, pitch high 2 second OSD menu go to opened, I can move up left down bottom and change. But foxeer monster v2 has 2 osd menus. General to make configs gain, brightness, color and so on and OSD for config runtime OSD - craft name, vbat, flight timer, and I can open only second OSD menu to config OSD - craft name, vbat, flight timer.

DieHertz, 2017-09-12:
From you description it's not clear what you did to get into the main menu, did you try yaw to the right?

csbuzkp, 2017-09-12:
Yes I tried do yaw high, pitch roll center. But main OSD do not show.

DieHertz, 2017-09-12:
Read the wiki page and try increasing key delay.

csbuzkp, 2017-09-12:
I will try once again all things to double check, I tried to change time from 100 to 220, step 20. Second menu work perfect. But some times main menu opened very rarely, but do not response by stick movements and closed spontaneously.

noypi2000, 2017-09-16:
I have the same issue with my Monster V2.  I have Omnibus F4 V2.  Used 220 resistor, 125 delay and 99 internal resistance.  I tried motor pin 5 and 6 with same result.  I can only go to the 1st menu like csbuzkp using joystick combo yaw low and pitch up.  If I change the internal resistance 50 using the same stick combo I can get to the 2nd menu but I cannot move up, down, left or right.

supiiik, 2017-09-19:
Same problem here with monster V2, omnibus F4 V1 (BF 3.2.0 RC5). joystick has standard resistors 0, 6.8, 15, 27, 45ohm, Voltages for keys are here:
up - 1,95V
down - 0V
left - 2,37V
right - 1,31V
Enter - 2,66V
From these values it seems, that internal resistance is 10,7kohm

I can't get 1st (need long press up) or 2nd menu (short press enter), but when I get it by the joystick, then I can move in 1st menu (name, voltage,...) all the ways and also confirm by the transmitter. In the second menu I can move only down (0V)
I was trying to change almost everything. The resistance seems to be correct at 107 (tested from 10 to 1000), 1st menu working properly from 75 to 140, so the middle value is also around 107. The delay is working from 100 to 500, but the higher number results in repeated menu movements. Value 100 seems to be fine.
I'm using 300ohm resistor without capacitor at A08 hardware PWM.
Is there something else, what I can try?

DieHertz, 2017-09-19:
I'll get a monster camera soon and see what can be done.

supiiik, 2017-09-19:
Found the problem. 1uF capacitor fixed it even at hardware PWM.
My settings for foxeer monster V2:

camera_control_mode = HARDWARE_PWM
camera_control_ref_voltage = 330
camera_control_key_delay = 125
camera_control_internal_resistance = 105

Don't forget the capacitor :)

supiiik, 2017-09-19:
It seems, that this camera has 2 HW independent OSD. One was working fine without capacitor and second one maybe need clean analog signal. By the logic analyzer the BF is sending 650kHz PWM, 1nF was not enough, so I've changed for the recommended 1uF and it's working.
One more thing. I can see on the analyzer, that the output has key delay 150 - 250 ms, but I have there camera_control_key_delay = 125

dannyfromnj, 2017-09-22:
I have a CL Racing F4 Flight Controller running BTFL, version: 3.2.0

When I enter "set camera_control_internal_resistance = 105"

I get "# set camera_control_internal_resistance = 105
Invalid name"

What am I missing?

mikeller, 2017-09-22:
@dannyfromnj: What version (RC) of 3.2.0 are you running?

dannyfromnj, 2017-09-22:
I'm running build 1633... downloading 3.2.0-RC5 Build 1​66​5 now.

DieHertz, 2017-09-22:
@supiiik 1uF is recommended for software PWM, there's no capacitor recommendation for hardware PWM AFAIK.
I hope I'll get a sample camera next week and just take it apart, no need to wonder anymore.

noypi2000, 2017-10-04:
I did what supiik suggested and it works for me too. Thanks for sharing.

closed 2017-10-04 by DieHertz

csbuzkp, 2017-10-04:
@DieHertz why you close this issue? you promised that it must work without capacitor.

DieHertz, 2017-10-04:
It won't. Redirect your request to Foxeer.

csbuzkp, 2017-10-04:
Хозяин - барин