betaflight/betaflight #4137: BF 3.2 RC4-5 stop FC working with PC

MimaRaza, 2017-09-12:
1.) SP Racing F3 firmware 3.2 RC5 after flashing directly connecting and working on it, configure it, without problem (Did it Sonday morning.) Maiden on Sonday afternoon without problems. 

Now wanna try change some settings i cant connect. Have the same issue with an Racerstar StarF3 board (i though its broken, but its the same problem) Shown both on com3.

2.) i tryed some ways to connect
    with ZADIG
    ImpulsRC Driver FIXER
    STM32Flashloader
    new driver
    restarting PC
   and so on

3.) SP Racing F3 works normally on the field, but i cant configure it now.

4.) Racerstar StatF3 cant test cause i cant configure it.;

5.)cant  Create a `diff` cause i cant connect!

mikeller, 2017-09-12:
Hi @MimaRaza. GitHub is for discussion and questions around development of Betaflight. For questions about your hardware, please contact the vendor, or check on RCGroups: https://www.rcgroups.com/forums/search.php?searchid=62321138&query=betaflight%20connect

closed 2017-09-12 by mikeller

richard-scott, 2017-09-12:
Firstly check you have the latest BetaFlight Configurator (BFC) installed.

Secondly, you aren't able to brick an FC by flashing firmware to it. The only way to make it not work in BFC would be if you a) disable USB-VPC or b) have a device you have soldered/connected to the board that conflicts with the USB socket working.

Even then its not bricked as you can re-flash by manually shorting the boot pins before connecting to USB.

asdiasx, 2017-09-12:
Aparently this version only works with msp on uart1...