betaflight/betaflight #4138: Downloads from ci.betaflight.tech partially broken

richard-scott, 2017-09-12:
I've looked on ci.betaflight.tech and I'm unable to download a .hex file as I used to be able to via the "[You can find the latest binarys here](https://ci.betaflight.tech/job/Betaflight/lastSuccessfulBuild/artifact/obj/)" link.

However, I have just noticed that if you dig into a [build ](https://ci.betaflight.tech/job/Betaflight/188/artifact/obj/)you can download the artifacts from the [.hex](https://ci.betaflight.tech/job/Betaflight/188/artifact/obj/betaflight_3.2.0_AFROMINI-188-b1f47c8a8.hex) links in there.

[image]

wind0r, 2017-09-12:
Thanks for the report :) Jenkins somehow crashed thats the reason for the error :(
Hex downloads still worked since they where cached by the CDN.

It should work now! Thanks again :)

closed 2017-09-13 by jflyper

richard-scott, 2017-09-13:
It's done it again ;-(

wind0r, 2017-09-13:
Could you clear the Browser Cache and try again?
For me and a friend it is working fine.