betaflight/betaflight #4142: rxfail request

peteoz, 2017-09-13:
would it be OK to request a feature whereby one of the failsafe options would allow to set rxfail 3 (throttle) to a specific value ?

for control channels only "a" (auto) and "h" (hold) are allowed, while AUX channels can be set to a specific value.

I can think of situations where setting rxfail 3 (throttle) to a specific value would be desirable.
I fly mostly in complicated terrain (ie. mountains) and anyway there seem to be already enough people preferring Betaflight for their long range setups (simple setups - no GPS) where in some situations it's not exactly easy (being far away) to judge correctly whether the pilot is just about to go out of LOS and loose a signal (typically behind a ridge, cliff, tall mountain,...).
in this situation it would be desirable (in case of failsafe) to automatically put the quad in auto-level mode and set throttle to a pre-determined value, with an expectation that the quad would rise up from behind the ridge/cliff in order to regain the signal.

we can already setup the failsafe to auto-level, but cannot set throttle to anything other than auto or hold. auto being zero throttle value and hold whatever was the last position, but in case of a descent behind a cliff/ridge the throttle is typically zero or low so putting it in to hold mode is not going to lead to a recovery.

I don't see any harm in allowing specific throttle settings for failsafe, keeping in mind that "hold" is already allowed (so potentially someone may have a failsafe at throttle value 2000 and allowing it to set to 1400 for example is not more dangerous).

it would also be preferable if there was a simple boolean settings for auto-level (on or off) in failsafe options, rather than wasting additional AUX channel for that (understanding that auto-level ON would only be available if accelerometer was enabled).

Arakon, 2017-09-13:
You can set the throttle level in failsafe when you select "Land" as stage 2 option.

peteoz, 2017-09-13:
but that's not really solution, stage 2 settings is for us to set up the way we want to bring the quad down when the signal is not recovered during guard time of stage 1.

stage 2 throttle value I'm using is actually low to ensure the quad comes down if stage 1 recovery failed.

jflyper, 2017-09-13:
So, you are saying that you want to be able to set throttle to some value so that it will gain some altitude while in stage 1?

Wouldn't that cause an armed craft, in any state (spinning out of control, resting on the ground at idle speed, etc.), to spin up to the value if radio is shutdown? It might make bloody faces if not bloody fingers...?

peteoz, 2017-09-13:
didn't think of that scenario but obviously to implement it correctly an additional check would be required (ie. active only if the idle throttle value was already passed).

apart from the fact that only users that really want the feature would actually make use of it.

DanNixon, 2017-09-13:
"apart from the fact that only users that really want the feature would actually make use of it."

You are obviously not familiar with how users work.

I think that might be OK with the condition that the present value is only applied when the throttle is above min_check (although this is also possible in flight). A check that throttle has been above min_check at least once is not sufficient to be safe IMO.

I have my reservations about this. I think there is too much scope to be in an unsafe situation if it is misused.

mikeller, 2017-09-13:
I agree with @DanNixon on this.

@peteoz: Essentially, what you are proposing is an 'autonomous flight' feature. 'failsafe' is a safety feature. Mixing safety features with other features is not the way to go, as this will reduce the safety aspect of the safety feature, and increase the chance of the safety feature not working due to misconfiguration.

Autonomous flight is not a focus in Betaflight, if you are interested in it then you should really go for iNav, there is a lot of development and testing effort being put into this there.

Out of interest, what are the reasons you are not using iNav for this type of flying? I am using iNav, and I find it is well done, and has a good feature set that is well suited for what you are describing. (i.e. autonomous navigation capabilities that allow for fine-grained configuration of how to react in RX loss situations).

peteoz, 2017-09-13:
@mikeller iNav is not for everybody, I'm not interested in autonomous flights, or flying from A to B at high altitude. I like to fly freestyle in the mountains, in complicated terrain, proximity, through the gaps, near the waterfalls and so on, for that I need a performance quad. while I have one 6inch quad, all of my other long range quads are 5inch, same quad I take for fun to a park or bando. I just prefer to stick with Betaflight, especially now with the 3.2 released with so many new exciting features.

why many Betaflight pilots don't want to move over to iNav for long range flights was already covered here :
https://github.com/betaflight/betaflight/issues/2714

if I flash my quads with iNav and start asking developers for performance features, I'll be referred straight back here. anyway, I'm not going to put GPS, baro on my quads so it's actually pointless going to iNav.

no big deal, I thought it was worth asking. btw. I don't see it as "autonomous flight" feature because we are talking about something that would last no longer than "guard time", in my case one second (although if implemented I would be tempted to extend my guard time to 2 seconds)

yes, failsafe is a safety feature which is divided in 2 steps - Stage1 (recovery attempt within guard time), Stage2 (bring the quad down if there is no recovery within Stage1). in the scenario I described above a quick vertical punch out would give a better shot at successful recovery during Stage1, that's all. but OK understand your reasoning.

mikeller, 2017-09-13:
@peteoz:
> btw. I don't see it as "autonomous flight" feature because we are talking about something that would last no longer than "guard time"

Fair enough. My point was not about it being an 'autonomous flight' feature per se, but about it **not** being a safety feature.

The trend in Betaflight seems to be going towards making it suitable for less technical users, and as part of this, even abandoing the concept of a configurable second stage and instead making the craft drop after the guard time has been discussed.

peteoz, 2017-09-13:
oh dear, that's gonna be few broken quads and gopros. they could remove 2nd stage settings from the configurator and leave it in CLI so that less technical people don't get to it easily.

I have my 2nd stage set to "Land" with throttle 1045 and during both failsafes I had last month my quads landed nicely on the belly (all my long range quads running top mounted battery). can send you videos if it helps the case.

closed 2017-09-14 by mikeller

DanNixon, 2017-09-14:
I don't think the issue is with your suggestion being useful in your case.

The issue I have is that outside of your narrow use case such a feature could pose a safety issue if users do not understand it and configure it properly (and people just do not take the time to do this).

Take for instance a quad sitting on the ground, armed with the throttle only just above idle (to circumvent a single check for throttle above min_check). What would happen then if a user has configured this feature and a failsafe even occurs?

If you can come up with a robust way of detecting when the throttle should go to the preset value and when it should remain at idle then it could be considered.

peteoz, 2017-09-14:
@DanNixon , I understand the suggestion I gave (as an example only) is not a robust way to safely detect when the event should be triggered. I'm not a developer and not technically minded to give valid suggestions, but I'm probably right to think that others who are involved in the development and have all the experience would be capable of coming up with a solid/safe detection, I'm just not one of those people.
if I was, I would probably modify and compile my own version to try.

I reviewed some of my DVR recordings with OSD where I use throttle value to see what throttle position I'm typically at during (nearly vertical) descent from some high peaks in the mountains, and contradictory to my earlier statement above (where I said I would be at zero throttle or near zero), it's actually not correct, during descent I'm on average around 30% throttle position, going as low as 20% but not lower.
with that in mind and taking into consideration all the valid safety points raised by other on this thread, if I was able myself to modify the code and compile FW for myself I would probably make a condition that a) accelerometer must be enabled and b) at the time of failsafe event my throttle position must be higher than 1200 (20%) as I'm assuming any quad with current throttle value of 1200 is probably already airborne and not sitting on the ground (but again I could be wrong or I might have omitted some other possibility).