betaflight/betaflight #4143: VBat not working RC5 SPF3

RobinFPV, 2017-09-13:
Hello, I upgraded my SPF3 to BetaFlight 3.2.0 RC4 and everything worked so far, including Vbat, so I updated my friend´s Quad to BetaFlight 3.2.0 RC5 since RC5 was out, but on his SPF3 VBat didn´t work anymore, after checking and resoldering all connections, downgrade to 3.1.7 and it worked fine, sorry I cannot follow all steps, but I don´t know how to fix it and stuff like that, just wanted to tell you.

RobinFPV.

jflyper, 2017-09-13:
@RobinFPV 
Battery meter on SPF3 do work with 3.2.0-RC5.
Please try full-chip erase flashing or do CLI "defaults" command and immediately test the feature.
If the issue persists, please redirect your support request to appropriate places on the net.

Ref:
https://www.rcgroups.com/forums/showthread.php?2387038-SPRacingF3-Racing-board
https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread

closed 2017-09-13 by jflyper

RobinFPV, 2017-09-13:
I did a full chip erase and the Settings we're stock.