betaflight/betaflight #4144: OSD Problem with CPU Overclocked

churd83, 2017-09-14:
The OSD flickers and has "-" (dashes) across the screen when Overclocking the CPU to try and run 32/16 on my Kakute F4 AIO. If I disable OSD it flies fine without and video issues. If I set CPU overclock off and set to 16/16 the issue isn't present either. Has to be something with the overclocking of the CPU. Tried to search this issue but haven't seen any post on it so I thought I'd throw it out there to see if this is known or not.

jflyper, 2017-09-14:
@churd83 
I believe the issue is triggered by SPI clock that is too fast:
- SPI clock for video overlay chip has been raised by fourfold (x4), as it was mistakenly halved from target value, which is half the rated max for the overlay chip. This brought the clock close to rated max on F4 (10MHz).
- When overclocked, with the current SPI clock manipulation logic, the SPI clock will increase in proportion, making it go beyond the rated max (goes from 10.5MHz to 15MHz).

I will look into this.

jflyper, 2017-09-14:
@DieHertz 
It bit us! 😉 
Interesting thing is that AB/AT7456 seem to hold at 15MHz, while Kakute F4, as I see in pics on the net, uses the original MAX7456. I suspect MAX, probably driven at 5V, is not capable of handling 15MHz clock, as oppose to AB/AT driven at 3V3.

churd83, 2017-09-14:
@jflyper
Thanks for looking into this. I don't understand the details but I understand your comment as a whole lol. So can this be fixed with an update, or because of the hardware used, I will not be able to run 32k on this FC with osd?

jflyper, 2017-09-14:
Possible solutions
1. Change the divisor handling to half the clock when overclocking is in effect.
Transparent to all bus devices, but slows down transactions considerably. This will hurt the gyro sampling also; if current code takes 7us to read, then it will take 14us to read, probably invalidating the original purpose of overclocking.

2. Half the default clock for overlay chip.
Transparent to all MAX/AT boards, but will run the chip at half the speed when not overclocked.

3. Half the clock for overlay chip for KAKUTEF4 target.
It will fix the KAKUTEF4 case, but there are clones of other targets that uses MAX and out-of-standard chips, which may develop the symptom.

4. Detect overclocking in overlay chip initialization and half the clock, possibly with a config variable that triggers this action, so users of AT/AB can enjoy less busy waiting.

Thoughts?

EDIT
The penalty imposed by slower SPI for MAX/AT will go away if DMA is in effect. I'm working on this (there is a proposed PR).

jflyper, 2017-09-14:
@churd83 I will build a test firmware that halves the clock by default. So we can test our hypothesis of clock being too fast.

churd83, 2017-09-14:
@jflyper awesome and thanks again.

jflyper, 2017-09-14:
@churd83 Looking forward to hearing the test results 😉

churd83, 2017-09-14:
@jflyper I will test tomorrow! Will report back. Thanks again

churd83, 2017-09-14:
@jflyper That WORKED! OSD isn't flickering anymore and I dont have the black dashes either. Will test fly and see if any other issues arise. I set it to 32/16 so lets see how that goes

churd83, 2017-09-14:
Flying great! Thanks again @jflyper!

closed 2017-09-14 by churd83

jflyper, 2017-09-14:
@churd83 Thanks for testing the PoC firmware. There will be an official (?) PR shortly. I might ask you to test it again 😄

jflyper, 2017-09-15:
@churd83 
It's going to be nice if you can test this firmware.
[betaflight_3.2.0_KAKUTEF4-4150-00.hex.zip](https://github.com/betaflight/betaflight/files/1305129/betaflight_3.2.0_KAKUTEF4-4150-00.hex.zip)

I'm interested in debug variables (you can observe them in sensor tab). Can you tell me what the values are under:
(1) Non-overclocked environment
(2) Overclocked environment

Thanks in advance.

churd83, 2017-09-15:
Yes I can test it tomorrow

churd83, 2017-09-15:
All debug are 0 for overclocked and non-overclocked

jflyper, 2017-09-16:
There must be something wrong... `debug[2]` should be 8 or 16.

`debug[0]` = Over clocking state (1 for overclocked)
`debug[1]` = Device type (0 for MAX, 1 for AT)
`debug[2]` = SPI clock divisor

Try this
[betaflight_3.2.0_KAKUTEF4-4150-01.hex.zip](https://github.com/betaflight/betaflight/files/1308030/betaflight_3.2.0_KAKUTEF4-4150-01.hex.zip)

mikeller, 2017-09-16:
@jflyper: Doesn't @churd83 need to set `debug_mode` to a particular setting for this to work?

jflyper, 2017-09-16:
@mikeller No, it's naive `debug` variable assignment.

churd83, 2017-09-16:
Overclocked
[image]

Non Overclocked 
[image]

jflyper, 2017-09-17:
@churd83 Yeap. Those are expected behaviors. The firmware works?

jflyper, 2017-09-17:
This is AB7456E (`debug[1]` is 1) under overclocking (`debug[0]` is 1).
The divisor (`debug[2]` remains at 8.

[image]

jflyper, 2017-09-19:
@churd83 
> Yeap. Those are expected behaviors. The firmware works?

I will give a go for fix merge as soon as I get a good test result.

churd83, 2017-09-19:
@jflyper yes I've put a handful of packs through this build without any issues. Goggles are broke at the moment and all I got is some cheap eachine one that I can't stand to fly with so I'm parked right now. Everything seemed fine to me without and bugs or glitches. OSD was clear so I'd say it's a go

churd83, 2017-09-19:
Thanks again for the quick response and fix!

jflyper, 2017-09-20:
@churd83 Ack! Hope you get your goggles fixed soon 😁

MylesShannon, 2017-09-21:
@jflyper 
The first hex you posted (betaflight_3.2.0_KAKUTEF4-max_half_clock.hex.zip) is the only working solution for the OSD rolling flicker I'm seeing while having "cpu_overclock=ON". Your next two subsequent hex files did not work with my FC in fixing this issue.

jflyper, 2017-09-21:
@MylesShannon 
If your board is influenced by the base cause of the issue reported in the first comment, the overlay will be entirely clobbered, not just rolling or flickering.

If you actually have the issue that appears only when overclocking is on, then I believe you should open the new issue ticket. In the ticket, a video clip that explains what "rolling flicker" looks like would be highly appreciated.

PR #4150 (2017-09-20): MAX7456: Handle SPI clock for CPU over clocked environment
PR status: Ready to merge
Fixes: #4144 

It was reported that Kakute F4, using legacy MAX7456 chip does not behave correctly under over clocked environment. With an assumption of the chip didn't like SPI clock going up from 10.5MHz to 15MHz, a test firmware that halves the SPI clock was created and proved the assumption to be correct.

Further investigation revealed following conditions to consider for a fix.

- MAX7456 does not work with 15MHz, while AT/AB7456 does (dependency on Vdd=5V/3V3 is not verified).
- Most newer designs use A [... cut]