betaflight/betaflight #4145: tasks CLI command gives very high % values for SERIAL task

0crap, 2017-09-14:
The first time after power on I enter CLI 'tasks' command, I get a correct value for the SERIAL task.
Every following 'tasks' command gives crazy high % values for the SERIAL task.

It's reproducable by going out of the CLI (click on Blackbox or whatever icon in BF Configurator) and click again on CLI icon.
First 'tasks' command is OK, every following crazy high % SERIAL task.
(NAZE target, BF3.2.0-RC5)

`05 - (         SERIAL)     99     498       3    5.4%    0.5%         7`
`05 - (         SERIAL)     96   50204       3  482.4%    0.5%        59`
`05 - (         SERIAL)     97   50201     113  487.4%    1.5%      1658`

jflyper, 2017-09-14:
@0crap 
It is natural that the SERIAL task shows unreasonable values, as CLI output blocks the task and remain it in the run state for prelonged period of time. It has no impact in flying.

closed 2017-09-14 by jflyper

0crap, 2017-09-14:
Thx for your reply @jflyper 
(Sorry for wasting your time. )