betaflight/betaflight #4147: VTX_COMMON needed by CMS when compiling

McGiverGim, 2017-09-14:
I don't know for sure if this is an issue or not. I think that each define is more or less independent to build customized versions or remove features to free some space if needed.
In an earlier version of Betaflight, I could add the CMS system without problem, but in the newest version there're some compile errors because it has a lot of references to some code that is included inside the VTX_COMMON.

Is mandatory the VTX_COMMON if I add the CMS? If it is, then I will close this issue.

How to reproduce? For example, here is target.h for a NAZE32 that adds GPS and CMS removing a lot of other features (not used by me). It now throws compile errors:

https://pastebin.com/t0urdEqu

In the code, I've added this:
```
#define CMS
//#define VTX_COMMON // Uncomment to solve errors
#define USE_MSP_DISPLAYPORT
```
Adding the define VTX_COMMON solves the problem and the NAZE32 hex is generated.

It seems to me that the CMS has a lot of references to the VTX_COMMON, so my C level don't let me analyse and solve the issue for my own.

Thanks!

DanNixon, 2017-09-14:
It shouldn't be, VTX configuration is just a small part of the CMS. There must be parts of the VTX config related code that are not properly guarded by `VTX_COMMON`.

I'll take a look later tonight if I get a chance.

jflyper, 2017-09-14:
Definitely not. Because CMS can work on external serially-connected OSD or I2C-connected OLED display alone. I must have done a loose job when I did the `VTX_COMMON` work... 😭

McGiverGim, 2017-09-14:
I've tried to reproduce it too with my OMINUBUSF4SD, adding this:
```
#undef VTX_COMMON
```
makes the compiler process fail. Maybe is a easier way to reproduce it.

PR #4148 (2017-09-14): Fix define dependencies for VTX code
Ensures CMS menus are only compiled if the required device driver is also compiled and removes compilation of device drivers and common code if either `VTX_COMMON` or `VTX_CONTROL` are undefined.

Fixes #4147
commit 811a78c5 (2017-09-14): Fix define dependencies for VTX code

Fixes #4147
closed 2017-09-14 by martinbudden