betaflight/betaflight #4149: Serial Receiver Provider can't be saved

NicolasAz1, 2017-09-14:
Hi,
Summary : 
Setting erased while connecting to BFC, and impossible to save back again.

Details : 
Flew my quad pretty well today, until I connected it to BFC today, changed settings on OSD, and Battery voltage alert... Then saved and realized several other things had changed : RX Serial receiver provider was no more "Ibus" but "Spektrum"... Tried to change, save and reboot, but after reboot "ibus" was replaced by "spektrum" instead.
As other settings still save well, I suspect it's not hardware-related ; and as I strictly observed ports setting as per user manual (see my config' in attachment), I think it's not config' related either. Maybe it's a little bug, but very annoying as my quad's grounded from now :-(.
Have tested with many other settings; all of them can be saved (OSD, motor settings, ports config, etc...) --> saving actually works. But for this one parameter "Serial Receiver provider" won't save ...

Versions : 
Drone model : Eachine Lizard95, Flysky receiver (ibus). Tx : I6. 
Betaflight 3.2.1. run on a Mac (Mac Os 10.7.5)
Below screen shot of port settings and value I'm trying to set and save for RX "Serial Receiver provider"

Arakon, 2017-09-14:
- You don't mention the betaflight version on the actual copter.
- As a workaround, try setting it in the CLI and saving there.

jflyper, 2017-09-15:
I suspect the BFC was automagically upgraded to the version not compatible with the running BF firmware version.

NicolasAz1, 2017-09-18:
Many thanks for your answers.
After browsing the web, I've found I was not the only one with this problem (https://multi-voltige.fr/viewtopic.php?f=25&t=4241) (in French). Still, as told by this user, I re-flashed BetaFlight (3.1, Dec 2016) and it worked ok… until I disconnected/reconnected, and the setting disappeared. I didn't have to flash a second time as saving via CLI (full dump) did seem to set receiver back to desired "IBUS" value… 
I close the issue as it seems there's a workaround. 
FYI : I flashed 3.2 on another lizard (I have 2 of them), and didn't have the same problem (yet ;-)).
Have a nice day.
Nicolas

closed 2017-09-18 by NicolasAz1

mikeller, 2017-09-19:
Not sure what the article says, my French is a bit rusty. Have you tried dumping the config, resetting the flight controller to defaults, and then restoring the config?