betaflight/betaflight #4151: My odd can't count... (Betaflight 3.2)

Bugg69s, 2017-09-15:
I have omnibus f3. Lipo: infinity 4s
At osd it is showing 16.8 battery voltage but at cell voltage it is saying 3.3 low battery. I think betaflight thinks that i have 5s lipo. When i connect and disconnect lipo sometimes like 5 times it is working. But at full voltage it is right. Can you add future to betflight that you can choose how many cells you have if you need to.

mikeller, 2017-09-15:
1. calibrate your voltage sensor: https://github.com/cleanflight/cleanflight/blob/master/docs/Battery.md#adc-sensor 
2. if 1. does not fix it. adjust 'Maximum Cell Voltage' on the 'Power & Battery' tab until it works (particularly for LiHV batteries).

robert-hoffmann, 2017-09-15:
I had this problem when i set max voltage to 4.2, and the battery was at 4.25 ..so i set it back to 4.3 and all was fine again.

Max bat MUST be inferieor to max voltage
..though this sounds like a bug kinda too. BF should just consider voltage=max voltage if voltage > max voltage

Bugg69s, 2017-09-15:
Thanks :) but I think it will be nice to can set how many cells you have, it will be cool if you will be able to set it through osd too.

Bugg69s, 2017-09-15:
I'm just waiting when they ad Command line to OSD!

mikeller, 2017-09-16:
>..though this sounds like a bug kinda too. BF should just consider voltage=max voltage if voltage > max voltage

The problem is that max voltage is what is used to calculate the number of cells. So if your battery voltage per cell is more than the max voltage, the logic will recognise the battery as one cell larger than what it really is.

AndrewFarley, 2017-10-23:
+1 would like this.  I've ran into similar issues with a few of my quads when using HV batts, and tried adjusting the maximum cell voltage and it still complains that I'm low voltage, assuming it's detecting one more cell than it should.  It would _really_ be nice to be able to (optionally) set what cell type, as many/most people only fly a single cell count type on a quad.

My current workaround is, on a HV batt plugging in power and flying for like 10 seconds while it's beeping and flashing at me in the OSD annoyingly, then landing, and power-cycling it, which then has the proper voltage and cell count without the annoying alarms.  😄

timman2er, 2018-09-25:
should  be closed via https://github.com/betaflight/betaflight/pull/6648

PaulFPV, 2022-02-16:
in #6648 I wrote this, maybe here is more visible:

I just experienced this problem on a long range quad (6S li-ion battery).

When the voltage dropped too much, the cell count went to 5S, reporting a totally wrong cell voltage.

So, a cli to set a fixed cell count is a must for long range users.

...I don't understand the state of this request... is there a way... or not.. to set manually the cell count at this moment?

klutvott123, 2022-02-16:
@PaulFPV According to the PR you linked to you should be able to do ```set force_battery_cell_count = "the number of cells you want"```

haslinghuis, 2022-02-16:
Fixed in #6648

closed 2022-02-16 by haslinghuis