betaflight/betaflight #4152: Motor reversing (anti-turtle) not working on BF3.2 rc5

frizzle101101, 2017-09-15:
Motor reversing not working on BF3.2 rc5 with speedex es 30a escs(dshot capable blhelis). arm with normal direction and then switch the switch and motors continue with normal direction. Arm with reversed direction switch and motors do not spin, but if you go to motor tab after and spin motors and they backwards untill you arm with switch not on reversed. Using 16.63 Dshot commands also tried 16.66 from the github. All done with a matekf405 and pdb combo fc (not aio).

mikeller, 2017-09-15:
- reversing while armed is not supported;
- in RC5, when the crash flip mode is activated, motors will not spin after arming when in default stick position. use pitch / roll to spin 2 motors at a time.

frizzle101101, 2017-09-15:
works perfect then, does it say that any where about the feature

brycedjohnson, 2017-09-15:
Recently the wiki has been updated with a short description.  https://github.com/betaflight/betaflight/wiki/BetaFlight-3.2

RamFPV, 2017-09-15:
Is small angle bypassed in turtle mode?

brycedjohnson, 2017-09-15:
@RamFPV No it isn't.

mikeller, 2017-09-16:
@brycedjohnson: Sure? I thought I was able to arm in turtel mode when upside down. And if it isn't, it probably should, because it kind of defeats the purpose otherwise.

brycedjohnson, 2017-09-16:
@mikeller I didn't touch any of the arming conditions, so I haven't tested it.  Your acc isn't off or small_angle set to 180?

brycedjohnson, 2017-09-16:
And yeah it should probably allow arming at any angle when in turtle mode.  I hadn't thought of that until now, because I have been running small_angle=180 for so long

mikeller, 2017-09-16:
Yeah, tested this today, with small angle at default you can't arm in anti turtle mode when upside down. Do you want to submit a fix (prolly just bypass angle arming condition detection when in crash flip mode), or should I?

RamFPV, 2017-09-17:
Yea same here tried it this morning. I'll just program a double switch arming sequence and turn small angle to 180 unless it gets bypassed in the future.

RamFPV, 2017-09-17:
Just got on github so if you could that would be great. Just learning how to use this site. Thanks!

fiveangle, 2017-09-17:
Just came here to submit an issue regarding the requirement to disable small_angle safety in order to use turtle mode.  As turtle mode is directly enabled, bypass of small_angle safety feature when mode is enabled would be safest/easiest.

mikeller, 2017-09-17:
@brycedjohnson: Are you working on this? Happy to fix, but don't want to double up on work.

brycedjohnson, 2017-09-17:
@mikeller I can put in the fix!

mikeller, 2017-09-17:
Thanks @brycedjohnson!

brycedjohnson, 2017-09-17:
Fix here: #4166

closed 2017-09-19 by mikeller

phillipsj70, 2017-10-24:
I still can't arm in turtle mode. I have 32bit  ESC's running with Dshot600. Dshot1200 gives to much notice right now. When my drone flips over. I disarm, switch to turtle mode, then rearm. Motors arm, but don't spin, until I move stick, but motor still doesn't reverse. Also the beeper not working. not sure if some how they can be related.

wind0r, 2017-10-24:
Thats totaly sounds like your esc's do not understand dshot commands. Check for updates or ask the manufacture to provide a update.
Nothing betaflight can do there.