betaflight/betaflight #4155: feature for osd elements on a switch - OSD profiles

frizzle101101, 2017-09-15:
was wondering if there is a feture for switching on elements of the osd with a switch, not just osd on off. like i have my default for racing but then i have another config i wanna use for freestyle with more osd info. All i wanna do is use a tx switch to switch these configs, sorta like rate profiles. Osd profiles i guess lol

DanNixon, 2017-09-15:
That is planned.

I have a demo of it working with @fishpepper 's OSD changes, but those will need to be finished first.

tigert, 2017-09-22:
We probably should also do something about the default layouts then - I filed #4211 as it is kind of a separate issue, but related, thus referencing it here.

pkruger, 2018-09-07:
I have a version of this working, just need to clean it up a little, for review and pull request.

mikeller, 2018-12-05:
Fixed by #6714.

closed 2018-12-05 by mikeller