betaflight/betaflight #4156: 3.2 including RC5 - LEDstrip no longer working on BluejayF4

ctzsnooze, 2017-09-16:
Hi guys, in upgrading from 3.1.7 to 3.2 RC4 and RC5, the LEDstrip has become inactive on my Bluejay F4 R4 boards.  

Curiously when using BLHeli to flash ESCs, the LED strip lights up, and it also glows weakly when plugged in to USB.  But it won't work at all when booting up from LiPo. 

It's a minor thing but I'd be grateful if there was a workaround to test.

Thanks - Chris

jflyper, 2017-09-16:
@ctzsnooze 
Did you do full-erase?

mikeller, 2017-09-16:
Curious. I flashed latest from master onto one of my quads with a BLUEJAYF4 and an LED strip, and it's working fine. How is your strip connected?

ctzsnooze, 2017-09-16:
Thanks...

Yes,  I did a full erase, several times to be sure.  And re-configured the settings rather than cutting and pasting.  

I've carefully re-checked everything.  The LED works fine in 3.1.7.  I have it set to a colour with warnings overlay and it flashes etc all as expected.  

But when flashed to 3.2 RC5, and then powered up from LiPo, the LED is always off.  It definitely has 5.03V power on it, and that voltage is the same as when booted into 3.1.7, so it's not a power issue.

Interestingly, if USB is connected and Betaflight active, and I then apply a LiPo, the LED will glow, though not the assigned colour.  And it also glows when using BLHeli to flash.  

Here is a diff dump, nothing odd that I can see:

`# diff
# version
# Betaflight / BLUEJAYF4 (BJF4) 3.2.0 Sep 11 2017 / 06:30:40 (649ccd7c7) MSP API: 1.36
# name
# resources
# mixer
# servo
# servo mix
# feature
feature MOTOR_STOP
feature TELEMETRY
feature LED_STRIP
feature ANTI_GRAVITY
feature DYNAMIC_FILTER
# beeper
# map
map TAER1234
# serial
serial 2 32 115200 57600 0 115200
# led
led 0 0,0::CW:2
led 1 1,0::CW:6
led 2 2,0::CW:6
led 3 3,0::CW:2
# color
# mode_color
# aux
aux 0 0 0 1300 2100
aux 1 26 1 1300 2100
aux 2 28 0 1700 2100
# adjrange
adjrange 0 0 3 900 2100 12 3
# rxrange
# vtx
# rxfail
# master
set gyro_lowpass_hz = 100
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 125
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 160
set moron_threshold = 16
set baro_hardware = NONE
set min_check = 1020
set rc_interp_ch = RPYT
set max_aux_channels = 8
set min_throttle = 1060
set min_command = 975
set motor_pwm_protocol = MULTISHOT
set yaw_motors_reversed = ON
set small_angle = 180
set auto_disarm_delay = 0
set gyro_cal_on_first_arm = ON
set pid_process_denom = 1
set frsky_vfas_precision = 1
set report_cell_voltage = ON
set sdcard_dma = ON
# profile
profile 0
`

mikeller, 2017-09-16:
@ctzsnooze: I just tried your LED settings with a build from master from  about 5 days ago, and it's all sweet (this is with an OMNIBUSF4, since I don't have a BLUEJAYF4 here):

https://photos.app.goo.gl/bV8at8Yhkr72oRgp1

As unlikely as it sounds, I'd try adding an in-line diode as described here: https://github.com/cleanflight/cleanflight/blob/master/docs/LedStrip.md#connections

ctzsnooze, 2017-09-16:
Thanks for confirming the issue is not generic to 3.2 or to F4 boards generally.  I don't have the problem on F3 boards either.

I won't try a diode since it works perfectly under 3.1.7 without a diode on two quads with these Bluejay boards, and fails on both of them on flashing 3.2.  Something has changed with 3.2.  

I note that the Bluejay resource listing is different in 3.2 from 3.1.7.  Perhaps these changes have something to do with it.

mikeller, 2017-09-17:
Considering you are driving your hardware outside of the specs, you should not be surprised if it does not work.

Any minor change in timing introduced by code changes could be driving your LED over the edge of the tolerance range, or so could a change in the level of the 5V or 3.3V rails, induced by changing current draw from any of the connected peripherals. Your description certainly fits this.

jflyper, 2017-09-17:
@ctzsnooze 
I think it's partly my bad.

It's a known issue, and there is a PR that addresses it, but is not scheduled for 3.2 at this moment. Let me think what I can do for you.

In the mean time, you can try remapping
```
resource led_strip b3
```
and use DEBUG pad for the LED strip signal (confirmed working with an emulated BlueJayF4).

jflyper, 2017-09-17:
Oh wait... You have some strange issue with how the board is powered? That suggest something different... weird.

AndersHoglund, 2017-09-17:
@ctzsnooze , I can confirm your issues on BluejayF4. With 3.1.7 a LEDstrip on motor 5 output works just fine. With 3.2.0-RC5 or current master HEAD (4f2bea6a848037872a81c165cd9073463db3059d) it fails. Testing using the default config, only LED-strip enabled and selected a red Larson scanner LED pattern. All LEDs are more or less bright white, with a faint wandering pink-ish pattern resembling the Larson scan. Same issue regardless if USB is connected or not. FC is powered by a 2S LiPo via a linear 5V regulator. I also have a series diode in seies on the LED-strip power line.

AndersHoglund, 2017-09-17:
Made a comparison test on REVO with 3.2.0-RC5, using the same LED-strip and power supply. Larson scanner works just fine on motor 5 output. 
Difference between BJF4 and REVO is motor 5 mapping, A01 on REVO and B00 on BJF4.

AndersHoglund, 2017-09-17:
Re-opeing issue, as this still looks like a real BJF4 problem.

AndersHoglund, 2017-09-17:
And here are the config used to show the problem.

```
# diff

# version
# Betaflight / BLUEJAYF4 (BJF4) 3.2.0 Sep 15 2017 / 14:02:41 (4f2bea6) MSP API: 1.36

# name

# resources

# mixer

# servo

# servo mix

# feature
feature LED_STRIP

# beeper

# map

# serial

# led
led 0 4,15::CO:2
led 1 5,15::CO:2
led 2 6,15::CO:2
led 3 7,15::CO:2
led 4 8,15::CO:2
led 5 9,15::CO:2
led 6 10,15::CO:2
led 7 11,15::CO:2

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master

# profile
profile 0

# rateprofile
rateprofile 0

# 
```

ctzsnooze, 2017-09-17:
Michael, there is nothing strange about how both these boards are powered.  The Bluejay F4 does not have an on-board 5V regulator, but has holes designed to mount a Pololu 5V regulator on the underside - I guess that appeoach is the 'recommended' solution.  I used a 600mA Pololu on each board. I only have one single programmable LED in the LED string.  Total measured current drawn is well below the Pololu limit.  Positive voltage at the LED is solid at 5.03V whether on or off.  It is precisely the same 5.03V whether running betaflight 3.17 or 3.2, it just doesn't work on 3.2.

Is remapping to b3 and using the debug pad likely to work?

jflyper, 2017-09-17:
Like I commented above, I think this is related with a N-Channel issue. I'm looking into it.

jflyper, 2017-09-17:
Okay folks, I've confirmed #3993 does fix the issue. Tested with BLUEJAYF4 emulation.

The problem is, we can not apply #3993 alone, as it is a part of big "Let's get `TIMER_OUTPUT_INVERTED` used correctly" campaign.

Let me think if we I can fix the LED strip on N-Channel alone.

ctzsnooze, 2017-09-17:
All I can say is thank you so much for everything you're doing.  Very gratefully appreciated.

mikeller, 2017-09-17:
@AndersHoglund: Thanks for confirming. I did a reset to defaults on my BLUEJAYF4, and now I get the same symptoms on my LED strip. Looks like I must have had some left over resource assignments from the old config.

@ctzsnooze: I didn't say it was strange, I said it was out of spec. With a Vin for the LED strip of 5V, the signal level of 3.3V provided by the MCU is less than the 0.7 * Vin that the WS2811 [specification](https://cdn-shop.adafruit.com/datasheets/WS2811.pdf) states for a valid 1 level. Most LED modules seem to be tolerant to being run out of spec for this, but it is touch and go, and minor changes in either of the voltages can drop them over the edge.
I have learned the hard way, after having LED modules stop working on me after changing some other components, so I make it a habit now to add the in-line diode to Vin on all builds with LED strip.
Your problem description (different behaviour depending on how the board is powered) certainly suggests that power levels play a part in this.

jflyper, 2017-09-18:
Since BLUEJAYF4 and CLRACINGF4 are the sole targets with LED strip on N-channel, I will temporary skip the `MOTOR_OUTPUT_INVERTED` handling in N-channel case.

This is the firmware with this hack.

[betaflight_3.2.0_BLUEJAYF4-3993_00.hex.zip](https://github.com/betaflight/betaflight/files/1309596/betaflight_3.2.0_BLUEJAYF4-3993_00.hex.zip)

ctzsnooze, 2017-09-18:
Problem fixed - flashed my RC5 dump over that build and the LED's are back on, working fine, colours flashing the lot.  Thank you so much!  :-)

AndersHoglund, 2017-09-18:
OK, great. Will test it via sources on Github and PR #3993. Question, is that PR now standing on it's own? Or still depending on the bigger n-channel re-work?

jflyper, 2017-09-18:
@ctzsnooze @AndersHoglund  Thanks for testing!!!

@AndersHoglund PR #3993 is now standing on it's own (with a hack), and can be merged toward 3.2 release.

MrSatoV, 2017-10-02:
Before I even try the hacked hex, is there a fix that's going in on next RC or other release - soon?  I have the CLRacingF4.  In fact, I'm owner of a shop that is selling these and need to know what to tell my customers - not just "here's a hack" because we want to sell only good and valid products and needing a hack doesn't look good.  It's fine for a short term until an update is done, I just need to know what I can tell them for when that will be.

MrSatoV, 2017-10-02:
Oh, and does "I will temporary skip the MOTOR_OUTPUT_INVERTED handling in N-channel case" mean I can not use reversed motors AND the LED strip with this hack?  And you say temporarily, so does that mean the official fix will allow for reversed motors AND LED strip?
I'm just curious too... why are these the only two targets who use N-Channel for this?  Is there a benefit over the usual method(s)?

ctzsnooze, 2017-10-02:
I fly reversed motors with this code and its not a problem.

jflyper, 2017-10-02:
@MrSatoV Shawn? Hey, long time no see! This is teralift@rcg.

There is a long story behind this issue.

- Inverted channels (aka N-channels) are timer channels that gives reverse polarity outputs. It simply behaves as if an inverter was applied to its output. It has nothing to do with which way a motor connected to this channel output rotates. Timer channels with this characteristic are sometimes explicitly marked with `TIMER_OUTPUT_N_CHANNEL` flag in source codes.

- There is another timer output characteristic flag called `TIMER_OUTPUT_INVERTED`. This flag is used when there is actually an inline physical inverter. Some board design with output driver circuit, for example KISSFC, that effectively inverts the polarity of the signal, need this flag. It also has nothing to do with the way motor rotates.

- There has been bugs in drivers that handles N-channels; motor driver over compensate the output to produce inverted polarity output; it only need to flip it once to get a normal polarity output, but it is flipping twice. LED strip driver couldn't even handle it correctly.

- There are many targets that use N-channels for motors. These targets pragmatically dealt with (double) inverted output from N-channel by applying `TIMER_OUTPUT_INVERTED` again (triple inversion!!!), which is not correct at all.

- This bug has been identified recently, when I reworked the BLUEJAYF4 timer channel assignments to resolve conflicts among channels and enable HEXA-Dshot. Some investigation work followed, and fixes were made during 3.2 dev & RC phase.

- Fixes, including motor and LED strip drivers, has been prepared, but since they require fundamental changes in timer channel definition file for so many targets, they were postponed until 3.3.

- Unfortunately, BLUEJAYF4's new (and CLRACINGF4's) mapping for LED was left untouched; not working at that time. There was a workaround to use the DEBUG pad (for BJF4), as described in the comment above.

- Then this issue was raised, and I thought I can fix this right now, by intensionally "double inverting" the N-Channel output. It would affect only two targets, BLUEJAYF4 and CLRACINGF4, which can be considered as isolated change and brought in even during the RC phase. The code injects a bug, hence I called it "_a hack_".

- In 3.3, this hack will be removed, and all drivers and targets dealing with N-channels will receive a complete overhaul.

I hope this story helps you to explain things to your customers.

ctzsnooze, 2017-10-02:
You guys are just so awesome!

MrSatoV, 2017-10-05:
Hey @jflyper aka teralift!  XD  Howzit!?
Wow, that's an awesome explanation!  Although, wonder how long before 3.3 is released.  o.O
Also, I just realized (right as I was about to flash) that that's the BJF4 target, not CLF4.  I don't suppose that one works for CLF4 as well?  If not, would it be possible to get one for CLF4?
DOMOOOOO!!!  XD

jflyper, 2017-10-21:
@MrSatoV Gees, I overlooked your comment.

BJF4 and CLF4 are different, and should use the different target firmware.
CLF4 should be fixed with the "_hack_", as you might have already found out.

closed 2017-10-21 by jflyper