betaflight/betaflight #4157: FLIGHT MODE SWITCH - CHANGING VTX CHANNEL (not intended)

lukenicholls9074, 2017-09-16:
Just setup my new quad on beta flight first time I've been using beta flight as I've switched from librepilotd (so cut me some slack as it might be something very simple). My problem is that my assigned flight mode switch is also changing my video transmitter channel, its not changing the band just flicks from say 8 to 6 to 2 randomly and so on.

mikeller, 2017-09-16:
@lukenicholls9074: GitHub is for discussions and questions around the development of Betaflight. For questions about hardware and configuration of Betaflight, please contact the manufacturer of your hardware, or have a look on [RCGroups](https://www.rcgroups.com/forums/index.php).

syperlaxer, 2017-11-30:
I am having the same issue whenever i change from a different flight mode i notice my channel goes up by one