betaflight/betaflight #4160: Flip32 F4 Airbot F4 Target

johnmoto1, 2017-09-17:
went from 3.17 to 3.2 rc4 and leds no longer work , have them on motor pad 5 , i also went to
and tried rc5 and still dont work , i have a block of 16 and only 2 led are on barely ,
changed changed to two other brand new leds and still no joy . 
as soon as you power on the battery they dont work .
tried reconfigure them in led setup and save reboot still nothing.
 flight controler is a flip32 f4 using Airbot f4 target.
leds connected to motor pad 5 using the signal plus and minus pads.
\\\
# resources
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 A01
resource MOTOR 6 A08
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B14
resource PWM 1 B14
resource PWM 2 B15
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource INVERTER 1 C00
resource INVERTER 2 NONE
resource INVERTER 3 NONE
resource INVERTER 4 NONE
resource INVERTER 5 NONE
resource INVERTER 6 NONE
resource INVERTER 7 NONE
resource INVERTER 8 NONE
resource INVERTER 9 NONE
resource INVERTER 10 NONE
[... 503 lines cut ...]
set gps_pos_p = 15
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_expo_yaw = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 83
set pitch_srate = 83
set yaw_srate = 83
set tpa_rate = 10
set tpa_breakpoint = 1650 
\\\

jflyper, 2017-09-17:
@johnmoto1 
LED strip has been changed to MCU pin PB6, which is connected to designated LED signal pin on most Airbot/FLIP32 F4 boards. Try rewiring your LEDs to the pin.

If you choose to keep the current wiring, you can do so by entering the CLI command
```
resource motor 5 none
resource led_strip a1
save
```

closed 2017-09-17 by jflyper

johnmoto1, 2017-09-20:
I did try the the led pads as I thought they may of changed it back but did not work , I will try the motor input in cli thanks

johnmoto1, 2017-09-25:
The remapping worked a treat thank you for all your hard work 🤗