betaflight/betaflight #4162: Crash recovery always on?

darksonny, 2017-09-17:
Hi guys. 

I have a very strange thing going on here... 
I turned the crash recovery on and off to see what it does, was doing about 50 packs since then from RC3 to RC4. Accelerometer is on. 
The only thing I activated is set crash_recovery = on, checked with get crash if it's really on or off. 
Matek F405 and SP Racing F3 board same issue, Flysky, DYS ESCS... 

MAINLY when the pack was almost empty, doing fast rolls or flips.... SUDDENLY from one acro move to the other, the quad would kick itself back to horizon which was quite scary the first time. 

In 50 packs, maybe 5 times I had that issue out of nowhere. 

It's getting stranger... As I'm using an experimental feature, I thought just turn that thing off again, it saved my ass a couple of times, but I don't like getting flipped against my will. 
NOPE, get crash shows me it's off, just had this issue some days ago again. Wtf 😉

Please help or suggestions, 
Thanks a lot in advance to this great community.

Vidalcris, 2017-09-17:
"MAINLY when the pack was almost empty, doing fast rolls or flips.... SUDDENLY from one acro move to the other, the quad would kick itself back to horizon which was quite scary the first time."

You can increase crash_delay to avoid this problem
set crash_delay = 50
The quad will then wait the error to be more than 50ms long before activating crash recovery.
If you still got false detection during fast acro move then try to increase the value again.

darksonny, 2017-09-17:
OK thanks for the message, I'll try. But it's still unexplained why it's also doing it with crash recovery set to OFF....

Vidalcris, 2017-09-17:
Yeah, as i cannot explain why your crash recovery is always on.. i just try to help you with what i know ;)

darksonny, 2017-09-17:
That's already a lot, because I didn't know yet what kind of variables I should play around with 😉

martinbudden, 2017-09-17:
@darksonny , please provide a CLI `diff` output.

adamtfc87, 2017-09-25:
I also seem to have crash recovery on even with it disabled. Using LUXF4OSD firmware. I'll get a diff posted as soon as I get home.

closed 2018-04-21 by DieHertz