betaflight/betaflight #4167: Quad flips out with sept 17 master

brycedjohnson, 2017-09-18:
Did this (#4161) break something for none 3d mode?  RC5 works well.

Anyone else having issues?

```
# diff

# version
# Betaflight / LUXF4OSD (LUX4) 3.2.0 Sep 17 2017 / 22:25:33 (ce824a6) MSP API: 1.36

# name
name [\]^_

# resources
resource MOTOR 1 A02
resource MOTOR 4 B00

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper

# map

# serial
serial 0 2048 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1200 1650
aux 1 13 1 1700 2100
aux 2 26 2 1300 1725
aux 3 31 2 1700 2100
aux 4 35 1 1300 1700
aux 5 36 3 1700 2100

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_notch1_hz = 0
set gyro_notch1_cutoff = 0
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 0
set min_check = 1010
set max_check = 1990
set serialrx_provider = SBUS
set blackbox_p_ratio = 64
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 270
set vbat_max_cell_voltage = 44
set current_meter = NONE
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_vbat_pos = 384
set osd_rssi_pos = 40
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2454
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 2401
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 385
set osd_mah_drawn_pos = 353
set osd_craft_name_pos = 2440
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_avg_cell_voltage_pos = 2433
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 465
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 472
set osd_stat_max_spd = OFF
set osd_stat_min_rssi = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_endbatt = ON
set debug_mode = GYRO
set vcd_video_system = 2
set displayport_max7456_wht = 3

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_lowpass = 90
set dterm_notch_hz = 0
set dterm_notch_cutoff = 0
set anti_gravity_gain = 5000
set setpoint_relax_ratio = 50
set dterm_setpoint_weight = 150
set p_pitch = 50
set d_pitch = 25
set p_roll = 42
set i_roll = 50
set d_roll = 25

# rateprofile
rateprofile 0

set roll_srate = 76
set pitch_srate = 76
set yaw_srate = 76

# ```

mikeller, 2017-09-18:
Dammit, forgot to set the `motorOutputMixSign` for non-3D mode (and didn't test non-3D after adding that, my bad :-(). Hang in there, hotfix coming.

mikeller, 2017-09-19:
Did #4168 fix it?

Arakon, 2017-09-19:
Just hovered on today's current master (props reversed, non-3D) and it worked fine, so I'd say yes.

closed 2017-09-19 by mikeller