betaflight/betaflight #4170: Omnibus F7 lost connection to Wraith32 35a esc with rc5 update

SkylionFPV, 2017-09-18:
I've still got my copy of rc4 so I'm good but I wanted you guys to know that when I flashed my Omnibus F7 to 3.2rc5 I lost the second set of startup beeps and it won't arm I'm assuming that it means I lost connection with those escs to the motors with the update for some reason, they beep and arm just fine on rc4 I've flashed back and everything is fine but it's weird, I'm not going to be switching escs to find out if it is an esc problem but I figured I should atleast let you all know

jflyper, 2017-09-18:
@SkylionFPV If you are using Dshot, then it is a known issue. #4129

dovcharov, 2017-09-18:
Same problem with me, seems to be known issue. Your ticket will be closed.

closed 2017-09-18 by jflyper