betaflight/betaflight #4177: Emax F3 Femto downgrade from 3.2.0 RC5 bricked the FC

czipis, 2017-09-18:
I bought Emax F3 Femto
https://www.emaxmodel.com/download/F3_femto_manual_V1.2.pdf

It came with BF 3.0.1 or 3.1.0 - I'm not sure really. I immediately flashed it to 3.1.7. (SPRACINGF3EVO)
So far so good. Everything looked good.
Then I upgraded to 3.2.0 RC5 and after some poking about remapping motor4 output for DSHOT and then decided to downgrade back to 3.1.7.
After successful flashing USB device is not recognized anymore and it flashes with red LED. 2x long, 5x short (according to FAQ it should be FAILURE_INVALID_EEPROM_CONTENTS).
I'm able to flash again to 3.2.0. RC5, then it boots normally, but then it doesn't detect accelerometer/gyro.

Any help?

jflyper, 2017-09-18:
> After successful flashing USB device is not recognized anymore and it flashes with red LED. 2x long, 5x short (according to FAQ it should be FAILURE_INVALID_EEPROM_CONTENTS).

Failure code starts with fast flashing, followed by slow flashes.
Is yours (5 fast + 2 slow), repeated twice?

czipis, 2017-09-18:
makes sense that the error is FAILURE_MISSING_ACC
but I don't understand how I could smoked it on the table.

mikeller, 2017-09-18:
As a last resort, try a 'Reset to Defaults' to make sure it's not stuck settings.

czipis, 2017-09-18:
resetting to default doesn't make any change.

jflyper, 2017-09-19:
Sorry, bad things can happen. I think it's time to write to your distributor.

closed 2017-10-23 by jflyper