betaflight/betaflight #4178: RC5 no ppm TX signal received CLracing f4 fc

Bubbasanjohn, 2017-09-18:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://rcgroups.com/forums/showthread.php?t=2464844), [Slack](https://slack.betaflight.tech) or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1.) Describe your problem;
2.) Include ways to reproduce the problem;
3.) Provide information about your flightcontroller and other components including how they are connected/wired;
4.) Add the used configuration and firmware version;
5.) Create a `diff` and post it here in a code block. Put \`\`\` (three backticks) at the start and end of the `diff` block; and
6.) Remove this Text :).

mikeller, 2017-09-18:
@bubbasanjohn: Can you please add the information as requested above?

Bubbasanjohn, 2017-09-18:
New RC5 firmware gets no tx signal from ppm.  rc4 and lower is no problem.  PPM is set up as normal.  
here is my dump on RC5.  Board designer said there was no change in his target code and I should post here.  

Using Jeti DS-16, ppm positive, 20ms frame rate, PPM is computed (direct doesnt make a difference), rx is set for 8 channels.

# version
# Betaflight / CLRACINGF4 (CLR4) 3.2.0 Sep 11 2017 / 06:31:34 (649ccd7c7) MSP API: 1.36

# name
name -

# resources
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 B08
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B09
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 B08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource INVERTER 1 C00
resource INVERTER 2 NONE
resource INVERTER 3 NONE
resource INVERTER 4 NONE
resource INVERTER 5 NONE
resource INVERTER 6 NONE
resource INVERTER 7 NONE
resource INVERTER 8 NONE
resource INVERTER 9 NONE
[... 542 lines cut ...]
set horizon_tilt_expert_mode = OFF
set gps_pos_p = 15
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_expo_yaw = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

mikeller, 2017-09-18:
Can you please post the output of a `diff`?

Bubbasanjohn, 2017-09-18:
sorry not sure what that means

mikeller, 2017-09-18:
Type `diff` in the CLI, post the output.

Bubbasanjohn, 2017-09-18:
# diff

# version
# Betaflight / CLRACINGF4 (CLR4) 3.2.0 Sep 11 2017 / 06:31:34 (649ccd7c7) MSP API: 1.36

# name

# resources
resource MOTOR 5 NONE
resource PPM 1 B09
resource ESCSERIAL 1 NONE
resource CAMERA_CONTROL 1 NONE

# mixer

# servo

# servo mix

# feature
feature -RX_SERIAL
feature RX_PPM
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -ARMING_GPS_FIX
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -MULTI_BEEPS
beeper -ON_USB
beeper -BLACKBOX_ERASE

# map

# serial
serial 3 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master
set dshot_idle_value = 800
set motor_pwm_protocol = DSHOT600
set small_angle = 180

# profile
profile 0

# rateprofile
rateprofile 0

#

Bubbasanjohn, 2017-09-18:
here is a dump from rc-4
# dump

# version
# Betaflight / CLRACINGF4 (CLR4) 3.2.0 Aug 28 2017 / 11:47:49 (b2cd7294e)

# name
name -

# resources
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 B08
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B09
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 B08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource INVERTER 1 C00
resource INVERTER 2 NONE
resource INVERTER 3 NONE
resource INVERTER 4 NONE
resource INVERTER 5 NONE
resource INVERTER 6 NONE
resource INVERTER 7 NONE
resource INVERTER 8 NONE
resource INVERTER 9 NONE
resource INVERTER 10 NONE
resource INVERTER 11 NONE
[... 539 lines cut ...]
set horizon_tilt_expert_mode = OFF
set gps_pos_p = 15
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_expo_yaw = 0
set thr_mid = 0
set thr_expo = 40
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

Bubbasanjohn, 2017-09-18:
here is a diff after setting up the TX, cuz rc4 works heheh
'''# diff

# version
# Betaflight / CLRACINGF4 (CLR4) 3.2.0 Aug 28 2017 / 11:47:49 (b2cd7294e)

# name

# resources
resource PPM 1 B09
resource ESCSERIAL 1 NONE
resource CAMERA_CONTROL 1 NONE

# mixer

# servo

# servo mix

# feature
feature -RX_SERIAL
feature RX_PPM
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -ARMING_GPS_FIX
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -MULTI_BEEPS
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE

# map

# serial
serial 3 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1900 2050
aux 1 1 0 900 1300
aux 2 2 0 1300 1700
aux 3 13 2 900 1300
aux 4 35 3 900 1300

# adjrange

# rxrange

# vtx

# rxfail

# master
set fpv_mix_degrees = 30
set dshot_idle_value = 800
set motor_pwm_protocol = DSHOT600

# profile
profile 0

# rateprofile
rateprofile 0

set thr_mid = 0
set thr_expo = 40

# '''

mikeller, 2017-09-19:
@Bubbasanjohn: Can you try this firmware and see if PPM works: 

[betaflight_3.2.0_CLRACINGF4.zip](https://github.com/betaflight/betaflight/files/1312525/betaflight_3.2.0_CLRACINGF4.zip)

@bnn1044: I suspect that this is caused by your changes in https://github.com/betaflight/betaflight/commit/72ace4566ed7475521e8f44825ac29f35d554e9e. This should probably be rolled back if the fix above is confirmed.

@jflyper: I think you did the `TIM_USE_ANY` changes. Do you see how this could affect PPM in? (and why is Tim using all these things anyway? ;-))

Bubbasanjohn, 2017-09-19:
> @bnn1044: I suspect that this is caused by your changes in 72ace45. This should probably be rolled back if the fix above is confirmed.

Yes I will try first thing tomorrow morning.  Thank you. FYI I did flash a copy of rc4 and it works.  Think I already posted that. Though. 

Anyway. This community is fantastic!  Thank you. 

John Sarran
Marin Pasta Works
877-263-2332 x 101

bnn1044, 2017-09-19:
@mikeller I make the change before RC4. it work on RC4 with PPM remap.

jflyper, 2017-09-19:
It's #3991. I will fix it.

jflyper, 2017-09-19:
@Bubbasanjohn @bnn1044

Can you test this firmware (3.2.0-RC5 based)?  It has been tested on the emulated CLRACINGF4.

[betaflight_3.2.0_CLRACINGF4-4182_00.hex.zip](https://github.com/betaflight/betaflight/files/1312776/betaflight_3.2.0_CLRACINGF4-4182_00.hex.zip)

mikeller, 2017-09-19:
@bnn1044: I know. But I wasn't sure that there was going to be a fix for this before 3.2.0, so a revert of your changes would have been a workaround.

@jflyper: Thanks!

jflyper, 2017-09-19:
Closing via #4182

closed 2017-09-19 by jflyper

Bubbasanjohn, 2017-09-19:
Yes this firmware works when I remap the ppm to B09.

PR #4182 (2017-09-19): F3/F4 PPM & PWM: Do proper IOConfigGPIOAF in ppmRxInit and pwmRxInit
PR status: Ready to merge

Fixes #4178.

#3991 should have included this.