betaflight/betaflight #4180: Turtle mode with reversed prop direction enabled not working

carloscampos775, 2017-09-18:
When enabling turtle mode the props spin however, the thrust is produced in the wrong direction. The quad will always pull its self into the ground instead of pushing itself over. 

FC: BetaFlight F3
Firmware: 3.2.0 RC5
ESCs: blheli_32 
Prop direction: Revered

mikeller, 2017-09-19:
What version of BLHeli32 are you using? Try updating to the latest version from https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM, earlier versions did not support the reversal commands.

carloscampos775, 2017-09-19:
I was using BLHeli32 v32.1. Updating to v32.15 *Beta* corrected the turtle mode issue with prop direction.

Close issue

jflyper, 2017-09-19:
You can close it yourself 😉

closed 2017-09-19 by carloscampos775

AndersHoglund, 2017-09-19:
This dependency to BlHeli_S and _32 versions need to be clearly stated, in the BF release note at least. There are nothing in 3.2.0-RC5 notes telling this.