betaflight/betaflight #4184: BLHeli passthrough not working in RC5 or master for NAZE

McGiverGim, 2017-09-19:
The passthrough for BLHELI is not working in the latest master. It gives this error:
[image]

The NAZE32 has default values, nothing changed or configured. I've tested with the two latest versions of the BLHeliSuite. My ESCs are littlebee blheli 20A (**not** blheli_S).

I have made several tests:
- Omnibus F4 V3 Pro with latest master -> Works (with different ESCs)
- NAZE32 rev6 10dof with version 3.1.7 -> Works
- NAZE32 rev6 10dof with version 3.2.0 RC5 - > ERROR
- NAZE32 rev6 10dof with version 3.2.0 (master) - > ERROR

How to reproduce: install latest master on NAZE32 and then try to connect from the latest version of BLHeliSuite.

lichtl, 2017-09-19:
@McGiverGim BLHeli pass through has been dropped for 3.2 on NAZE target, because of the flash size constraints.
https://github.com/betaflight/betaflight/issues/3701

McGiverGim, 2017-09-19:
Ok, I didn't know that. Can someone point which define is the responsible of the passthrough? Maybe a custom compilation will let me add it. Thanks!

McGiverGim, 2017-09-19:
@mikeller point to me to this: USE_SERIAL_4WAY_BLHELI_INTERFACE
I will try and close this issue. Thanks!

closed 2017-09-19 by McGiverGim