betaflight/betaflight #4187: Glitchy throttle and roll signals after moving to BF 3.2 RC5

johnblackj, 2017-09-19:
Updated from RC4 to RC5 using configuration 3.2.2.
When I try to fly, I can hear the motors slow and then speed up again. I also experienced some issues then trying to do barrel roll. Sometime the roll pauses and then continues.
On RC4 it was flying pretty nice.

If I look at the receiver tab, I can see the value on the throttle signal glitch and drop momentarily and then return to the correct value.
I can get the roll to do the same but the behavior seems a little less frequent

I'm using a spektrum dsmx receiver. Only thing that changed was the version of betaflight

I tried going back to an older 3.2 hex I had saved but the problem is still there.

mikeller, 2017-09-19:
> I tried going back to an older 3.2 hex I had saved but the problem is still there.

This suggests that this wasn't introduced by your upgrade, but something that happened around the same time. Check your RX wiring.

johnblackj, 2017-09-19:
Will do,  I think the wiring is good as I can see the glitch in betaflight without anything running so there is no vibration to cause issues.
However I use set spektrum_sat_bind=7 to bind. And I can remember for the life of me if I use 9 the last time. 
Gotta retry when I get home.