betaflight/betaflight #4188: How to programme an automatic flip/roll/yaw in betaflight CLI ?

kinfpv, 2017-09-19:
Hello,

Almost all my question is in the title, I'm sorry if I didn't put it in the right place, but we can consider that not finding any answer to my a software question is an issue, isn't it ? ;) 

So, for a project I need to be able to program (in CLI ?) some specific task on my FPV drones (x220 builds, Omnibus F3). Can you tell me if it's possible with BF3.2 and where to search explanations about this kind of programmation ?

Thank you very much for your help and for this great software,

KIN

Arakon, 2017-09-19:
You don't. You'd have to code that function from scratch.
Betaflight isn't a scriptable autopilot.

kinfpv, 2017-09-19:
Hello Arakon, thanks for your answer. I'm asking how to programme it from scratch in Betaflight, I saw so many functions in the CLI that I can imagine it could be possible… If, as we know, some 15$ drones has this function I suppose it shouldn't be too hard to program, don't you think so ?

If it's seriously impossible to program this kind of movements in BF, then do you know any other firmware able to do that ?

Thanks in advance,
KIN

kinfpv, 2017-09-19:
Just to feed this post :
What about doing it with Open TX (using Taranis X9D+) ?
Or with "Swift Playgrounds" app maybe… ?

See you

Arakon, 2017-09-19:
CLI has nothing to do with programming. It just enables or sets things that were already programmed. Simply put, if you want to program a feature like that, you'll have to learn to code, understand the betaflight source code fully, and implement the function yourself. It's not hard to do most likely if you can code, but it's simply not a feature a lot of people will want to use.

mikeller, 2017-09-19:
@kinfpv: If you are interested in programmability in the flight controller, you should have a look at [iNav](https://github.com/iNavFlight/inav).

kinfpv, 2017-09-19:
@Arakon Thanks for your Answer, find attached a little quick photoshop I made for fun… ;)
But actually it could be interesting for multiple reasons : 
- a way to check if PIDs are correct (Stinger Swann method, quick hard change of directions)
- make super low tricks (close to the floor)
- make more complexe combinaisons of tricks
- create succession of tricks (like rubikscub for exemple), 
- learn how tricks looks like live to learn how to do it manually faster
- make better quality videos (perfect 360° rolls…)
- for small drone company who want to commercialise toy-drones with auto-roll/flip (as it exists)

I know that if a menu like this would appear it would be super awesome :) !
And probably not only for me I suppose…

@mikeller : thanks for sharing, I'll check it ;)

[image]

jflyper, 2017-09-20:
IMHO, it's going to be a cute addition, if someone can take care of all the regressions...

fiveangle, 2017-09-20:
hubsanflight, here we come !

-=dave

digitalentity, 2017-09-20:
Such kind of additions would be definitely nice to have.

kinfpv, 2017-09-20:
@digitalentity : I agree, and for sure it's future of FPV, next step for more complexe freestyle + better quality videos… In my opinion if we don't make BF go into it another firmware will do it anyway… GPS and other sensors are becoming more and more affordable, democratized and light weight, and we cannot just ignore the possibilities it offers for racing, freestyle and security / safety (return home), feedback/learning. Added to this, the distance sensors (sonars) and this kind of things.

An other amazing feature would be to add a "record and replay" mode on every mouvements of the drone : you fly a race or some freestyle, and when you stop it you put a new battery and you replay it to watch yourself flying without your googles… Like this ghost in a racing video games for exemple. Of course you can take back the control on it at anytime you want with the remote or a back to home option or a safety landing… Like a cruise control in a car (it's not really auto-pilote, it's better to save energy of the car and your own ability to focus).

Sorry if I'm thinking loud, but I need to put it somewhere ! I guess we need to have more creativity offered in programming for users of BF (it's highly exciting, so make people use it more, buy more parts, sensors, and make BF even more indispensable) ;)

Have a nice day you all,
KIN

mikeller, 2017-09-20:
@kinfpv: What you are describing in https://github.com/betaflight/betaflight/issues/4188#issuecomment-330837761 are pretty much all things that are the focus of what iNav does. The reason that Betaflight and iNav are different is because they want to be different. It's not that we are begrudging iNav for the users who use it because they want  things like GPS / MAG / Sonar / ... support and waypoint navigation (and more). And iNav does not begrudge us for the users who choose Betaflight for the ultra fast loop times and performance tuned PID controller (and more). In fact, most developers in here are also contributors in iNav, and there is a healthy transfer of functionality between the two. So if it's in iNav and not in Betaflight, it's safe to assume that this is because it was decided that Betaflight did not want it.

kinfpv, 2017-09-20:
Ok thanks for this informations, now I understand better. I see that creativity for programmers is in between acro (BF) and programmed functions (iNav) … and so it is (will be more and more) for users… in my opinion, the gap should get thinner. But I understand that for programming it could be confusing to mix both approaches from scratch. 

Many hardwares (FCs, ESCs) + fame/success of Betaflight make that people almost "has to" use it, or kiss (but it's more and more Betaflight). So many videos, tutorials etc. With my co-pilots we all use it now for exemple (including people doing models since 20 years). And I thought it was a contributive open-minded community, and I hope it is… I slitty feel irony in some answers to my questions, as if it would be stupid… I hope I'm wrong. 

I do freestyle and race and I'm confident enough, so be sure that it's not a question from a beginner. I'm not into this demonstration of performances and competition, based on fully manual use called acro… simply because it's already not like that at all : PIDs, Antigravity, Dynamic Notch Filters, Air modes, Turtle mode… For me the most interesting part it's how creative can become somebody getting into drones or who are into drones since long time.

So ok for iNav, but tell me if I'm wrong, I have to flash it on my FC first (it means erase my BF settings on an existing quad) isn't it ?

Have a nice day,
KIN

mikeller, 2017-09-20:
> So ok for iNav, but tell me if I'm wrong, I have to flash it on my FC first (it means erase my BF settings on an existing quad) isn't it ?

Yes. But you can easily create a backup of all of your settings (CLI tab, type `diff all`, Save to file), and then restore it to its original configuration if you are going back to Betaflight.

kinfpv, 2017-09-20:
@mikeller , Thanks, I knew it - about "diff all" - but thanks again for share the tip. 
And is it possible to it all the way around : `diff all` in CLI to save the settings in inav ?

digitalentity, 2017-09-20:
@kinfpv There is no way of saving settings to transfer them between Betaflight and INAV. In fact trying to do so may cause unexpected behavior. With INAV it's better to do a clean install (reset settings) and set up from scratch. One of the presets in Configurator will give you a decent start.

TBH for 90% of pilots out there properly set up INAV in ACRO mode feels exactly the same as properly set up Betaflight. The difference in so small that only a handful of hardcore pros can feel it. Please

kinfpv, 2017-09-20:
@digitalentity , thanks for sharing ;) I'm not asking if there is a way to transfer from BF to iNav, I just want to know in iNav if making a CLI `diff all` copy/past Back up works properly (saving, but mainly if it works when you past it back in the CLI) ;)

digitalentity, 2017-09-20:
@kinfpv yes, INAV adopted CLI `diff all` command from Betaflight.

TiagoTiago, 2019-12-31:
I'm not yet too familiar with the full extent of capabilities of Betaflight, but doesn't sound it by itself allows for such advanced automation.

But I think the auto-flip feature might be doable with some advanced OpenTX programming, possibly involving automated switching to and from Horizon mode and the overriding of some channels. This of course, would require you have a quad with Horizon mode set to a switch matching what you have in your transmitter programming.

Or if you don't mind losing access to the maximum deflection of the sticks and always flying in Horizon mode, then that's all you have to do; that's basically the deal with Horizon mode from what I understand, it's Angle mode with auto-flipping when you push the sticks all the way.

As for more complex tricks, to some extent you can sorta macro it with just OpenTX (once again, advanced trickery), and just fly the quad on Acro mode as usual; though, this will of course be dependent on the PIDs and rates of the quad, and would be susceptible to "desyncs" due to external forces and any tiny variations in the performance of the quad and stuff, and likely would only be dependable for quick rotation tricks, producing potentially unpredictable results for longer tricks and/or tricks involving changes in position, and small variations in the initial angle in relation to gravity as well as speed relative to the wind and/or obstacles and ground may introduce significant distortions (don't expect to be able to race a track automatically, and even powerloops and orbits may be unreliable).

tavdog, 2022-08-02:
close bump

closed 2022-08-03 by haslinghuis