betaflight/betaflight #4189: 3.2.0 RC4 - arming in horizon (or switching to horizon before arming) inverts motor directions

DEvil0000, 2017-09-19:
3.2.0 RC4; latest blheli: 16.67. F3 minicube
Arming in horizon (or switching to horizon before arming) inverts motor direction.
I have arming on a slider at aux 1 and acro/horizon on a slider at aux 2.
When initially setting up my quad with 3.2.0 RC3 (or RC2) I had "dshot reverse motors" also configured on aux 2 (like reverse/acro/horizon) but I removed it then after one test. So there is no reverse configured now.
I fly acro most of the time but sometimes when landing or taking off I switch to horizon for convenience. When pluggin in my battery I normally have the transmitter aux 2 position on acro but I am not 100% sure about that. However when arming in horizon (or switching to horizon and arming then) motors are inverted. Disarming, switching to acro, arming again sets the motors to correct rotation again. Switching to horizon while its armed works as expected.

jflyper, 2017-09-19:
@DEvil0000 Can you attach cli command "diff" AND "dump" outputs in separate comments?

DEvil0000, 2017-10-03:
# diff
# Betaflight / OMNIBUS (OMNI) 3.2.0 Aug 28 2017 / 12:01:58 (b2cd7294e)
resource MOTOR 1 B09
resource MOTOR 2 A02
resource MOTOR 4 B08
mixer QUADX1234

feature -RX_PPM
feature INFLIGHT_ACC_CAL
feature RX_SERIAL
feature MOTOR_STOP
feature RSSI_ADC
feature LED_STRIP
beeper -ARMING_GPS_FIX
beeper -RX_SET
beeper -MULTI_BEEPS
beeper -SYSTEM_INIT
beeper -ON_USB
serial 0 1 115200 57600 0 115200
serial 1 128 115200 57600 0 250000
serial 2 64 115200 57600 0 115200
led 0 0,15::CIW:0
led 1 7,15:S:CTW:2
led 2 8,15:S:CTW:2
led 3 15,15::CIW:0
color 1 0,255,80
color 2 0,0,80
color 3 30,0,80
color 4 60,0,80
color 5 90,0,80
color 6 120,0,80
color 7 150,0,80
color 8 180,0,80
color 9 210,0,80
color 10 240,0,80
color 11 270,0,80
color 12 300,0,80
color 13 330,0,80
aux 0 0 0 900 1400
aux 1 2 1 900 1650
aux 2 13 0 1825 2100
aux 3 27 0 1950 2100
aux 4 35 1 1475 1550
set gyro_lowpass_hz = 65
set gyro_notch1_cutoff = 225
set gyro_notch2_hz = 120
set gyro_notch2_cutoff = 70
set baro_hardware = NONE
set mid_rc = 1516
set min_check = 1120
set max_check = 1940
set rssi_channel = 9
set rssi_scale = 30
set serialrx_provider = SBUS
set rx_min_usec = 1000
set blackbox_p_denom = 64
set blackbox_device = SERIAL
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 15
set failsafe_kill_switch = ON
set align_board_yaw = -90
set vbat_warning_cell_voltage = 34
set current_meter = VIRTUAL
set vbat_scale = 111
set ibatv_scale = 400
set yaw_motors_reversed = ON
set deadband = 6
set yaw_deadband = 7
set osd_rssi_alarm = 60
set osd_alt_alarm = 30
set osd_tim1 = 2048
set osd_tim2 = 1537
set osd_vbat_pos = 2440
set osd_rssi_pos = 2146
set osd_tim_1_pos = 87
set osd_tim_2_pos = 2167
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_ah_sbar = 200
set osd_current_pos = 353
set osd_mah_drawn_pos = 321
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 2456
set osd_warnings_pos = 2345
set osd_avg_cell_voltage_pos = 399
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2154
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set camera_control_key_delay = 150
profile 0

set dterm_lowpass = 65
set dterm_notch_hz = 120
set dterm_notch_cutoff = 70
set setpoint_relax_ratio = 30
set dterm_setpoint_weight = 100
set yaw_lowpass = 90
set p_pitch = 24
set i_pitch = 16
set d_pitch = 7
set p_roll = 20
set i_roll = 16
set d_roll = 7
set p_yaw = 31
set i_yaw = 20
set d_yaw = 15
set p_level = 40
set i_level = 40
set level_sensitivity = 60
set level_limit = 60
rateprofile 1

set rc_rate = 90
set rc_rate_yaw = 90
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80

#

DEvil0000, 2017-10-03:
# dump
# Betaflight / OMNIBUS (OMNI) 3.2.0 Aug 28 2017 / 12:01:58 (b2cd7294e)
name -
resource BEEPER 1 C15
resource MOTOR 1 B09
resource MOTOR 2 A02
resource MOTOR 3 A03
resource MOTOR 4 B08
resource MOTOR 5 B07
resource MOTOR 6 B06
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B04
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource I2C_SCL 1 NONE
resource I2C_SCL 2 NONE
resource I2C_SDA 1 NONE
resource I2C_SDA 2 NONE
resource LED 1 B03
resource LED 2 NONE
resource LED 3 NONE
resource RX_BIND 1 NONE
resource RX_BIND_PLUG 1 NONE
resource TRANSPONDER 1 A08
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 NONE
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 NONE
[... 493 lines cut ...]
set p_level = 40
set i_level = 40
set d_level = 75
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_sensitivity = 60
set level_limit = 60
set horizon_tilt_effect = 75
set horizon_tilt_expert_mode = OFF
rateprofile 1

set rc_rate = 90
set rc_rate_yaw = 90
set rc_expo = 0
set rc_expo_yaw = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80
set tpa_rate = 10
set tpa_breakpoint = 1650

#

DEvil0000, 2017-10-03:
maybe the missing peace was: omnibus
https://www.banggood.com/Eachine-Minicube-F3-6DOF-Flight-Controller-Intergrated-Betaflight-OSD-20mm20mm-p-1126931.html?rmmds=search

jflyper, 2017-10-03:
@DEvil0000 Can you try RC6?

DEvil0000, 2017-10-03:
Maybe in 14days. This copters esc + fc just burned today.Now I do not have a working one.

jflyper, 2017-10-03:
Oh... sorry for your loss... 😭

mikeller, 2017-10-05:
Yeah, sorry for your loss. But looking at your `diff`, it appears that the flip after crash mode was still configured for AUX 2:

`aux 4 35 1 1475 1550`

DEvil0000, 2017-10-05:
It was definitely not visible in the configurator UI. Guess something went wrong removing it.That was somehow what I expected because noone else faced the bug and I had it comfigured once.

mikeller, 2017-10-05:
@DEvil0000: That is entirely possible. Sometimes the configuration gets corrupted during an upgrade. In these cases, backing up (in CLI), resetting to defaults, and restoring will fix it.

closed 2017-10-23 by jflyper