betaflight/betaflight #4209: Can not get the acceleration data by MSP

su497835388, 2017-09-22:
Board： SP RACING F3_EVO _Brush V2.0      STM32F303CCT6+MPU6500
Issues:   Use teensy3.2 to get the the board‘s status by MSP RX .   Useing same method, teensy3.2 can get "MSP_ATTITUDE" data normally,  the "MSP_RAW_IMU" data was wrong.

  uint8_t data = 0;
    sendMSPRequest(MSP_RAW_IMU, &data, 0); // Send MSP Request:

f3 respone: $ M > 12  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF

mikeller, 2017-09-22:
Seems to be working just fine here when I am testing with configurator. Have a look here for how it's used: https://github.com/betaflight/betaflight-configurator/blob/master/tabs/motors.js#L317-L338

su497835388, 2017-09-22:
@mikeller  So?  I  cannot get the acceleration data by teensy3.2 ?  you know  I am newbie

mikeller, 2017-09-22:
I can't see why not. The command seems to be working fine, and I have pointed you to a working reference implementation that you can use  to compare your code with, to see if there is anything that is wrong that stops it from working.

su497835388, 2017-09-22:
@mikeller Thank you so much

su497835388, 2017-09-25:
@mikeller hello sir  Do you kown the scale of the acceleration data ,I got the data Z aixs about 515 when F3 was static

mikeller, 2017-09-25:
https://github.com/betaflight/betaflight-configurator/blob/master/js/msp/MSPHelper.js#L109-L112

su497835388, 2017-09-25:
@mikeller thankes a lot O(∩_∩)O~

closed 2017-10-23 by jflyper