betaflight/betaflight #4211: Sensible defaults needed for OSD layout

tigert, 2017-09-22:
I think the current default OSD layout is pretty much a mess. While it might be useful for development to see everything at a glance :-) it is pretty laborious to set up a sensible layout for a new board.

https://www.instagram.com/p/BYK5HBdgRnC/

We should have a sensible, reasonably usable OSD out of the box.

For development one can have a cut&paste config dump to enable whatever you are hacking on, but this is totally different from the case when you actually want to have a flyable quad. I suggest we define something pretty minimal that works for a racing/freestyle quad as the defaults.

In the spirit of "patches speak louder than words" here's my config - I can make a proper pull request once we discuss this a bit - I might well be unaware of something, and my setup does not necessarily have everything a default setup should have.

Also my config is PAL - I guess defining something for NTSC is the lowest common denominator, as the elements will also fit on a PAL screen, although not optimally - but there is no way to know what kind of camera you'll have on your build.

[image]

set osd_vbat_pos = 2498
set osd_rssi_pos = 2489
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2518
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 2081
set osd_crosshairs = 200
set osd_ah_sbar = 200
set osd_ah_pos = 200
set osd_current_pos = 2507
set osd_mah_drawn_pos = 2103
set osd_craft_name_pos = 65
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 65
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 53
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 2466
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2379
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83

DanNixon, 2017-09-23:
I'll echo what I said in Slack here.

The best default IMO is for every element to be disabled and positioned in the centre of the screen.

The advantage of this being only configuring the elements you actually want, no layout suits everyone so it seems sensible to just let the user start from scratch (not all boards have a current sensor or barometer, so even on your reduced config people will be disabling elements).

This also leads to cleaner config diffs as the diff would only contain elements you actually use.

mikeller, 2017-09-24:
I am in favour of @DanNixon's suggestion. This has the additional advantage that it reduces the flash space that is required by the firmware to store the defaults.

In additon, I would like to see a (configurator only) 'Show All' button that enables all elements, and lays them out in a spread out fashion on the 'demo' screen, so that new users can see what's available at a glance.

robert-hoffmann, 2017-09-26:
I'm for something inbetween. Like just put voltage (i suppose 99% of everyone will want that anyways) center top, and warnings center bottom ..thats it.

If it where for me i'd ask for a different layout, but BF needs something that suits 95% of everyone out of the box ..so it should probably be just the most basic of information (ie warnings/voltage/time) ..or simply nothing at all.

mikeller, 2017-11-07:
@tigert: I think the 'simple but usable' out of the box (i.e. in the firmware) is taken care of sufficiently with #4260. betaflight/betaflight-configurator#686 should have the developers rejoice. And as you say, experienced users won't be using the OSD tab much anyway. The one thing that I think would be a 'Load sample config' button in configurator that would allow newbie users to load a reasonable set of presets.