betaflight/betaflight #4213: Dshot commands buzzer

Bugg69s, 2017-09-22:
I have tried to setup buzzer to beep when I failsafe but did not work. At Betaflight it showed that it should beep but it does not. (Omnibus f3, x6b receiver, ibus, fs i6, Betaflight 3.2 RC5)

tigert, 2017-09-22:
This is not a support channel, generally read/search/post to the rcgroups thread for usage questions. 

But by default the beeper should beep a few times when the board boots for example when connected to USB unless beeping has been disabled in the configuration. So it might be connected wrong if there are no beeps at all or the beeper might be of a wrong type - consult the manual of your board?

Bugg69s, 2017-09-23:
At boot up they beep, even when I set it to switch it beeps but when I set it to beep when I failsafe if doesn't beep

Bugg69s, 2017-09-23:
I'm using Dshot command beeper

jflyper, 2017-09-23:
Dshot beeper only beeps when initiated by AUX switch (BEEPER mode), intended to be used for lost model finder and not for general notification and alarm uses.

This is because the motor beeping is done by vibrating a motor back and forth very quickly; if you want the motor beeper to beep while the motor is spinning, motor has to be stopped, and that is usually not a very good thing to happen.

For further assistance, please redirect your questions to some other appropriate forums on the net.

Thank you in advance.

Bugg69s, 2017-09-24:
@jflyper i mean that when i crash and my quad disarm it seld and should start beeping but it doesnt beep.

Bugg69s, 2017-09-24:
Disarm it self

jflyper, 2017-09-24:
> Dshot beeper only beeps when initiated by an AUX switch (BEEPER mode), intended to be used for lost model finder, and NOT for general notifications and alarms.

Bugg69s, 2017-09-24:
@Jfliper I set it to when I failsafe it will beep, when I power up quad first is beeping and when I power up transmitter is stops- that's Ok.
And when I then set to any mode (like acro, air.. ) and turn off my transmitter it beeps thats Ok too.

But when I arm and then power off transmitter it disarm it self, set it self to beeper but doesn't beep at all, in Betaflight is shows that it should beep (in that slider menu when you change aux channels to flight modes, I now I can't remember name of it)

jflyper, 2017-09-24:
Are you trying to make the Dshot beeper beep when failsafe kicks in by setting failsafe value to AUX channel that is assigned to BEEPER mode?

Bugg69s, 2017-09-24:
Yes

Bugg69s, 2017-09-24:
@jflyper yes that's it. It beeps like I set it to as fast I dosn't have armed when failsafed

jflyper, 2017-09-25:
@Bugg69s 

Okay, here is what's happening.

- When failsafe kicks in, it sets the AUX channel that you have assigned to BEEPER for `RX_SET`.
- The failsafe also set the `RX_LOST` beeper.
- The `RX_LOST` beeper have precedence, so the beeper is beeping indicating `RX_LOSS`.
- Dshot beeper only works with `RX_SET`, so it doesn't beep. (It may tickle, but not beep.)
- If you have a real beeper, then it should be beeping with `RX_LOST` pattern.

mikeller, 2017-09-25:
One more point is that, even if the `RX_SET` beeper was working, the Dshot beacon would not be activated in case of a failsafe, since it only activates when the quad is disarmed.

jflyper, 2017-09-25:
@mikeller The craft disarms at the end of the failsafe process.

mikeller, 2017-09-25:
@jflyper: But I assume that at that point the stage 1 channel presets are not in force any more, so using them to activate the beeper won't work in any case.

I think it might be a reasonable approach to have the Dshot beacon activate at the point when the craft disarms during a failsafe (if the user has enabled Dshot beaconing).

jflyper, 2017-09-25:
> But I assume that at that point the stage 1 channel presets are not in force any more, so using them to activate the beeper won't work in any case.

Channel value set by failsafe remains after failsafe process and the craft is disarmed.

mikeller, 2017-09-25:
> Channel value set by failsafe remains after failsafe process and the craft is disarmed.

Ok. Then the Dshot beacon should kick in at that point, since it only checks for the mode being activated, and not for beeping priority.

jflyper, 2017-09-25:
@mikeller 
IMO, that requires the failsafe to be aware if `BEEPER_RX_SET` is in effect when it wants to set `BEEPER_RX_LOST`.

Bugg69s, 2017-09-25:
Ok. Is there way to turn Beeper_RX_lost off?

jflyper, 2017-09-25:
@Bugg69s No. Even if you stop RX_LOST from beeping, the condition still exists, and it will suppress the RX_SET from beeping.

mikeller, 2017-09-25:
@jflyper: Good point. Added `BEEPER_RX_LOST_LANDING` as an additional condition for the Dshot beacon: #4231.

@Bugg69s: Can you test #4231 please?

Bugg69s, 2017-09-25:
@mikeller test what ?

mikeller, 2017-09-25:
@Bugg69s: Test if #4231 works the way you want it for using the Dshot beacon in case of failsafe. If you are not set up to build the firmware yourself, I can build one for you if you let me know what target you need.

Bugg69s, 2017-09-26:
Ok

Bugg69s, 2017-09-26:
@mikeller where I need to build firmware ?

mikeller, 2017-09-26:
@Bugg69s: If you aren't set up to build the firmware already, it's easiest if you let me know what board version you use, and I'll build it for you.

Bugg69s, 2017-09-27:
@mikeller Ok, thanks I'm using omnibus f3 nonpro

mikeller, 2017-09-27:
Here you go:

[betaflight_3.2.0_OMNIBUS.zip](https://github.com/betaflight/betaflight/files/1338716/betaflight_3.2.0_OMNIBUS.zip)

closed 2017-10-14 by Bugg69s