betaflight/betaflight #4214: Add DAC for camera control already!

DieHertz, 2017-09-23:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://rcgroups.com/forums/showthread.php?t=2464844), [Slack](https://slack.betaflight.tech) or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1.) Describe your problem;
2.) Include ways to reproduce the problem;
3.) Provide information about your flightcontroller and other components including how they are connected/wired;
4.) Add the used configuration and firmware version;
5.) Create a `diff` and post it here in a code block. Put \`\`\` (three backticks) at the start and end of the `diff` block; and
6.) Remove this Text :).

DieHertz, 2017-09-23:
Anyone willing to test it may try one of these builds:
[betaflight_3.2.0_BLUEJAYF4.hex.zip](https://github.com/betaflight/betaflight/files/1326136/betaflight_3.2.0_BLUEJAYF4.hex.zip)
[betaflight_3.2.0_MATEKF405.hex.zip](https://github.com/betaflight/betaflight/files/1326137/betaflight_3.2.0_MATEKF405.hex.zip)
[betaflight_3.2.0_STM32F3DISCOVERY.hex.zip](https://github.com/betaflight/betaflight/files/1326138/betaflight_3.2.0_STM32F3DISCOVERY.hex.zip)

```
resource CAMERA_CONTROL A4 # or A5
set camera_control_mode = DAC
save
```

Some observations:
* At least with STM32F3DISCOVERY max DAC output is 3% lower than ADCref, so I had to bump `camera_control_ref_voltage` by 3-5%.
* You may shave 15-25 ms off `camera_control_key_delay` as DAC settles much faster than PWM + RC-filter
* DAC is either PA4 or PA5, on STM32F3DISCOVERY you can use either, on BJF4 and MATEKF4 only one of them is broken out, consult with your FC wiring diagram or use a multimeter in continuity mode

1337flite, 2017-09-23:
I'd be happy to test the Matek  - I've been trying to work out how to use the DAC for camera control but have not been able to works out the pin name.

MATEKSYS, 2017-09-23:
DAC is on PA4 pin of Matek F405 target

1337flite, 2017-09-23:
Matek F4 AIO  mainly works for me, thanks,
I can't seem to navigate down the menu i.e. pitch does not move the "cursor"/menu selection.
I've checked my endpoints for pitch and it looks OK (1000-2000), so not sure if you can normally use pitch to navigate the menu downwards, but I thought you could and can't with this version.

Other than that seems to be fine.

Thank you Diehertz - this feature rocks.

```
# version
# Betaflight / MATEKF405 (MKF4) 3.2.0 Sep 23 2017 / 02:43:10 (845b628) MSP API: 1.36

resource CAMERA_CONTROL 1 A04

# get camera_cont
camera_control_mode = DAC
Allowed values: HARDWARE_PWM, SOFTWARE_PWM, DAC

camera_control_ref_voltage = 330
Allowed range: 200 - 400

camera_control_key_delay = 180
Allowed range: 100 - 500

camera_control_internal_resistance = 470
Allowed range: 10 - 1000
```

1337flite, 2017-09-23:
OK the camera OSD works, but when I try to arm my quad the OSD warnings is showing "NOGRORO" and the quad will not arm.

I've tried this 3 times flashing between RC5 and this hex (inlcuding a fulll chip erase), and the problem persists.

As soon as I free the CAMERA_CONTROL resource i.e.:
```
resource CAMERA_CONTROL 1 none
save

```
I can reboot and arm the quad as nomrmal.

DieHertz, 2017-09-23:
Thank you for testing it, I'm looking into how the "no gyro" issue can be solved.

isilcala, 2017-09-26:
This is AWESOME! But I have a Runcam Eagle v1 on my MATEK F405 build which OSD has 5 signal pins! Not sure if there's an elegant way to support Eagle v1 as well? (And my next MATEK F405 build will go with a Runcam Split, oh MY!)

OtterNas3, 2017-10-02:
Function is working properly on Matek F405-AIO
However as 1337flite mentioned, arming get disabled with "nogyro"

Another thing i noticed is:
After
**resource camera_control 1 A04**

Fc is not able to connect to PC on USB anymore

after free the pin again and reboot of FC it works normal again.

OtterNas3, 2017-10-17:
@DieHertz 
Any updates on this?

DieHertz, 2017-10-17:
Not for 3.2, I'll prepare a PR for 3.3 out of my branch the next week

bhuism, 2017-11-23:
@DieHertz which of your branches has this?

OtterNas3, 2017-12-14:
@DieHertz do you still have the source of the hex you posted in the beginning of this thread?
Would like to have a look myself, to maybe sort the issues

brycedjohnson, 2017-12-15:
I'll give it a test on NERO if you have it!

mazatracker, 2018-02-11:
I can't quite get if this feature made it to 3.2.4 or 3.2.5 at least? Very interested in DAC pin as camera controller!

mikeller, 2018-02-11:
@mazatracker:

> mikeller modified the milestones: Betaflight v3.3, Betaflight v3.4 10 days ago

rosik89, 2018-03-13:
Really looking forward to this feature, my camera is already connected to the PA04 on the Matek F405 CTR. Is it sure that it will be implemented in BF3.4?

DieHertz, 2018-03-13:
As long as it stops conflicting with gyro, I hope it will :-)

rosik89, 2018-03-14:
@DieHertz ok, if you have a version to test just say it.

gzesiek, 2018-03-20:
@DieHertz : Hi, I would be also able to test it quickly (since I also soldered it to DAC pad not knowing it's not supported... not YET supported :smiley:) on MatekF405. I'll be watching for early-:bird: test-:floppy_disk:

eta232, 2018-03-20:
Whats The actually setting/wiring suggestion for matek f405 aio ctr with foxeer hs1117? Tried connect on dac PIN and enabled as camera control Mode,  but not sure how to set the correct pin if neccessary..

mazatracker, 2018-03-20:
Looking forward for this feature and could test it on Matek F405 with Arrow V3 and Monster V2 cameras, as well as Holybro Kakute F4 + Arrow V3

Diaoul, 2018-04-28:
How can I help making this happen? I can contribute code & test on Matek F722 with Runcam Eagle 2 Pro. The DAC pin is closer than S6 and no resource remapping would be required.

controlflow, 2018-06-21:
Still not in 3.4? :(

ijustwannafly, 2019-01-15:
Is there any development on this?  I have two matek f4 ctr boards with dac pin and cameras I would like to control.

mikeller, 2019-01-15:
@ijustwannafly: Not at the moment - the fact that DAC seems to have a hardware conflict with SPI looks like it could make this problematic.

erakli, 2019-06-03:
Could we expect updates on this feature in BF4? If there’s something in software part, I would like to help. 

It’s a must have feature because it push user experience with BF to new level.

mikeller, 2019-06-03:
@erakli: Can you explain a bit more what this 'must have feature because it push user experience with BF to new level' is, compared to using PWM for camera control?

Also, feel free to open a pull request for a fix to the hardware conflict that is blocking this at the moment.

erakli, 2019-06-04:
I think that plug’n’play behavior of camera control feature is really necessary, because there’s so many information on how to get things work that newcomers may become frustrated. For real, 20-minutes long video from Bardwell on how to make it happen – this is something over complicated for such small thing. 

Okay, I’ll see if I had enough competency to fix hardware problem.

jflyper, 2019-06-04:
DAC based camera control, even if implemented correctly, does not provide the plug’n play behavior by itself, as it is a mere substitution of PWM+RC filter based DAC.

jflyper, 2019-06-04:
DAC will add another few minutes to the video lol

erakli, 2019-06-04:
If it’s a PWM + RC, then we have a great replacement for soldering physical resistors and capacitors. this is more plug’n’play solution, than we have now. 

The other option is to buy cameras with UART, but I’d rather have more convenient options like DAC.

mikeller, 2019-06-04:
@erakli:

> If it’s a PWM + RC, then we have a great replacement for soldering physical resistors and capacitors. this is more plug’n’play solution, than we have now.

The RC elements can simply be added to the flight controller by the manufacturer, as some already do, making it just as 'plug and play' as the DAC solution.

erakli, 2019-06-04:
So, you think, it’s easier to solve the problem on hardware side? Is there huge difficulties to fix DAC in BF (to be honest, I haven’t investigate the cause of the conflicts yet)?

mikeller, 2019-06-05:
@erakli: Well, solving this on the firmware side is all down to solving the resource conflict there seems to be between DAC and SPI that is described earlier in this thread. Until this is done, solving it on the hardware side is easier.

marianomd, 2020-01-03:
For hexacopters this would be great, since it frees the motor output.

RipperDrone, 2020-01-03:
@marianomd - I typically hook camera control up to the LED pin and remap it to camera_control 1. Unless you want to use LEDs on your hexas, this gives you enough free motor outputs on 6-PWM ports FCs :-)

craigmunday, 2020-08-01:
Hi @DieHertz, Is there a pull request associated with this issue that could be reviewed?

mikeller, 2020-08-01:
@craigmunday: No. There still is no solution for the hardware conflict between SPI and DAC that is stopping this from working properly.

craigmunday, 2020-08-01:
I do have some draft code on the iNav fork which sends a 400hz triangle wave form via the DAC pad for the Matek F405 CTR and was also able to set voltage levels on this pad.  The fight controller also armed ok with no gyro warning.

I was wanting to review some of the earlier code written on this betaflight topic to see what differences there are. I didn't want to progress if I was going to run into the same roadblock that was experienced on the betaflight code

mikeller, 2020-08-02:
@craigmunday: That's good news. Not sure @DieHertz ever opened a pull request, as the hardware resource conflicts became apparent pretty early on, and to my knowledge were never solved.

nightflasher, 2020-09-13:
just wanted to ask: any news on this with the inav code?

closed 2026-09-26 by blckmn