betaflight/betaflight #4217: matek f405 OSD broken boot button

Spoonman61, 2017-09-23:
i have a matek f405 osd and the boot button fell off with normal pressure.  the switch has 4 contact points, front 2 are mounting points the back 2 activate the switch, i verified this with a multimeter.  it pulled the pads when it fell off so i have nothing to short across, does anyone know the location of traces i could short to?

jflyper, 2017-09-23:
@Spoonman61 The Betaflight git hub is a place for discussion towards better flight controller software. Please redirect your questions and support request to appropriate places on the net. Thank you in advance.

closed 2017-09-23 by jflyper

mikeller, 2017-09-23:
@Spoonman61: Also, this sounds very much like a situation where you should ask your supplier for a warranty replacement.

Spoonman61, 2017-09-24:
Franks guys, sorry about posting in the wrong forum.  I'll try to get a replacement.