betaflight/betaflight #4220: Not issue, just something that i will like :)

Bugg69s, 2017-09-24:
I whould like to have the boot from the SD card. Like when I insert SD card with Betaflight on it, it will boot Betaflight with my configuration and then I inserted SD card with like raceflight or inav and it will boot that software from the card with configuration, that will mean that on flight controller will be software that will need to read SD card and run software on it, but not  write software from sd card to it's storage. 
That will be awesome ! You just get 256mb SD cards and you could run any FC software without flashing. :)

jflyper, 2017-09-24:
@Bugg69s 
Interesting, but very hard, if not impossible.

This is because MCUs for small embedded systems, such as STM32 family we are currently using, are designed and optimized to run code on internal FLASH using internal RAM for variables.

ledvinap, 2017-09-24:
@jflyper : You can use SD-capable bootloader that will reflash MCU. It may make sense on F4 with large flash (with enough space for bootloader without significant optimizations)

jflyper, 2017-09-24:
@ledvinap I've given that method for a second, but it's not "without flashing" 😉

ledvinap, 2017-09-24:
@jflyper : It would behave as if everything is stored on SD (program + configuration). It may be useful for in-field testing.
And as for bootloader, it won't be that complicated - just verify program in flash (using checksum), compare it with version on SD (if sd is inserted) and flash blocks that are different if necessary.
Then either run main program (with checksum match), or blink failure.

Configuration storage may be implemented as stick commands - copy config to/from SD - this will be useful even without flashing part.

mikeller, 2017-09-24:
I am with @jflyper there. I would want to have the flashing part initiated with stick commands or OSD. Imagine accidentally overwriting your working firmware with some test that didn't work a month ago, just by reusing that sd card to write a log...

Also, this would open the opportunity to have multiple firmware versions on the SD card, and switching between them without having to fiddle out the card. (Betaflight / iNav combo anyone?)

Bugg69s, 2017-09-24:
I think it could work, memory on FC will be almost empty it will have just software to remap memory to sdcard. First it will try to boot from FC but flight controller will remap it to SD card and it will boot from SD card.

jflyper, 2017-09-24:
@Bugg69s 
We can't boot from SD card. We must first have to flash the internal flash with whatever you want to boot with. It's just there isn't enough RAM, and codes on SD card is not directly executable.

Btw, do we have data path constraints about the I-Bus on general RAM other than CCM/TCM?

Bugg69s, 2017-09-24:
@jflyper I don't really understand. Can you make it simpler for me? Im from Slovakia and translator don't really helped me understand

DieHertz, 2017-09-24:
@Bugg69s your idea can't with on an STM32, just forget it :)

jflyper, 2017-09-24:
@Bugg69s 

Take STM32F405RG for example. This is a very popular MCU that is used in majority of F4-based FCs.

This particular MCU has
- 1MB of FLASH memory
- 192KB of RAM

It is not like your PC, in which there is a separate block of RAM chips that provides giga bytes of memory.

If you want to execute a software stored in SD card with this MCU, the software must be read into either of these memories.

- Reading into FLASH is called flashing.
- Reading into RAM is program loading, but the RAM is so small that it can probably fit very restricted set of features.

martinbudden, 2017-09-24:
@jflyper we could implement virtual memory and demand paging... (joke)

Bugg69s, 2017-09-24:
@jflyper oh I see now. 192kb of ram is not enough. But I think it will be possible. If you made to on that memory is like: don't boot from me boot from pin x that is connected to SD card, it will make CPU think that SD card is its normal memory and just booted from card. I don't know if it is possible but it could be.

mikeller, 2017-09-24:
@martinbudden: Can we get it to email out statistics of the paging activity? ;-)

@Bugg69s: I think there is a part of what @jflyper tried to explain that you did not understand: There is a fundamental difference in the hardware configuration of your typical dedicated CPU chip (like the one in your PC for example), and 'all in one' MCUs like the STM32 family: The dedicated CPU has a set of pins that are used for the data and address bus, and a variable amount of RAM or ROM (i.e. flash storage) can be added externally to the CPU, and programs residing in either of these can be executed directly. On a (STM32) MCU, these busses are internal to the chip only, so anything has to be loaded into either flash or RAM before it can be executed. This means that there is no way that the CPU will ever 'think' that data on an SD card (or even an external flash chip) is executable directly, just the same as, on a PC, there is no way that the CPU can directly execute data that is on an USB stick (even if the operating system will pretend to do so, by loading the program into RAM in the background.

Bugg69s, 2017-09-25:
@martinbudden oh. Ok

closed 2017-09-25 by Bugg69s