betaflight/betaflight #4223: Onboard dataflash chip not found when OSD feature is disabled

Haasje, 2017-09-24:
Right now, my quad is running without VTX hardware. I'm just flying it LOS.
When I when to setup my quad in betaflight, I first flashed the latest firmware to get the nice DSHOT commands. After the flash I could see that the configurator had found the onboard flash chip for blackbox logging.
When I finished the configuration and rebooted the FC the flash chip would still be found, but when I power up the quad with the battery and then connect to it, the datachip is gone. This only seems to happen after I power up the quad with the lipo. As long as I leave the lipo disconnected and just power thru USB the datachip would stay active.

After a lot of reflashing and testing I found that this problem occurs when I disable de OSD in the configuration. I did this because for now I'm not flying with VTX hardware anyway. If I leave the OSD enabled, the dataflash chip also remains active.

I'm not sure if this is a bug or just an as-is, but I thought I would drop it here anyway.

Flightcontroller: Holybro Kakute F4
Betaflight version: Betaflight / KAKUTEF4 (KTV1) 3.2.0 Sep 11 2017 / 06:38:35 (649ccd7c7) MSP API: 1.36
Betaflight configurator: 3.2.2

\\\
# version
# Betaflight / KAKUTEF4 (KTV1) 3.2.0 Sep 11 2017 / 06:38:35 (649ccd7c7) MSP API: 1.36

# name
name quad

# resources

# mixer

# servo

# servo mix

# feature
feature AIRMODE
feature ANTI_GRAVITY

# beeper

# map
map TAER1234

# serial

# led

# color

# mode_color

# aux
aux 0 0 0 1925 2100
aux 1 1 1 1925 2100
aux 2 13 3 900 1075
aux 3 26 2 1400 1600
aux 4 31 2 900 1075
aux 5 35 1 900 1075

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set baro_hardware = NONE
set min_check = 1010
set max_check = 2000
set blackbox_p_denom = 48
set blackbox_device = SPIFLASH
set dshot_idle_value = 300
set motor_pwm_protocol = DSHOT600
set pid_process_denom = 1

# profile
profile 0

set dterm_lowpass_type = PT1

# rateprofile
rateprofile 0
///

mikeller, 2017-09-24:
Interesting find, @Haasje. Could you please post the output of a `resource list`, once with OSD enabled, and once more with OSD disabled?

Haasje, 2017-09-24:
This is the resource list with OSD disabled and the data chip not found by the configurator:

# resource list
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FLASH_CS
B04: LED 2
B05: LED 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: FREE
B11: SERIAL_RX 3
B12: FREE
B13: FREE
B14: SPI_PREINIT
B15: INVERTER 3
C00: FREE
C01: FREE
C02: FREE
C03: ADC_BATT
C04: MPU_CS
C05: MPU_EXTI
C06: FREE
C07: FREE
C08: FREE
C09: BEEPER
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: MOTOR 4
DMA1 Stream 2: MOTOR 2
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: MOTOR 3
DMA1 Stream 7: MOTOR 1
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

I then enabled the OSD again, rebooted, unplugged the usb, plugged in the lipo, plugged in the usb and connected the configurator again. The datachip was back. This is the resource list:

# resource list
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FLASH_CS
B04: LED 2
B05: LED 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: FREE
B11: SERIAL_RX 3
B12: FREE
B13: FREE
B14: OSD_CS
B15: INVERTER 3
C00: FREE
C01: FREE
C02: FREE
C03: ADC_BATT
C04: MPU_CS
C05: MPU_EXTI
C06: FREE
C07: FREE
C08: FREE
C09: BEEPER
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: MOTOR 4
DMA1 Stream 2: MOTOR 2
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: MOTOR 3
DMA1 Stream 7: MOTOR 1
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

jflyper, 2017-09-24:
The diff is
```
< B14: SPI_PREINIT
---
> B14: OSD_CS
```

Very interesting...

mikeller, 2017-09-24:
It is indeed. On KAKUTEF4, OSD and dataflash both are on SPI3. Could this be caused by the SPI speed for OSD being changed, resulting in the speed being too high for the dataflash?

Haasje, 2017-09-24:
@jflyper I just tested it. It has to to with the power cycle of the FC. When I disable OSD but don't unplug and just save and reboot, the datachip remains active. When I disconnect, unplug usb, plug in usb and reconnect then the datachip is gone. This is all without having the lipo on it.

jflyper, 2017-09-24:
Hmm... not likely, because modern flash chips are often capable of handling the ultra fast clock at over 100MHz, but I'll look into it.

Problem is that I don't have the Kakute...

jflyper, 2017-09-24:
> When I disable OSD but don't unplug and just save and reboot, the datachip remains active. When I disconnect, unplug usb, plug in usb and reconnect then the datachip is gone. This is all without having the lipo on it.

It's getting even more interesting !!!

Haasje, 2017-09-24:
The other way around, so enabling the OSD and the saving and rebooting brings back the dataflash right away, I don't need a power cycle then to bring it back.

mikeller, 2017-09-24:
Ok, I think this might be a case to contact your manufacturer about. I have tested your configuration on a KAKUTEF4, and my data flash was properly recognised and working when connecting the battery and then connecting USB, or connecting just USB.

Haasje, 2017-09-24:
did you turn the OSD option on and off too?

mikeller, 2017-09-24:
Scratch my previous comment, I had tested with the `diff` that you'd posted above. On reading your description for the second time I realised that this must have been the diff of the working configuration.

mikeller, 2017-09-24:
@jflyper: It does look indeed like we've got a problem with the dataflash initialisation. When using a version with OSD compiled out entirely, the dataflash chip remains unusable. Could it be that we are missing some part of the SPI init for dataflash?

jflyper, 2017-09-24:
@mikeller I think `spiPreInitCs` is not working for MAX for some reason.

mikeller, 2017-09-24:
For MAX? Not rather for the dataflash?

jflyper, 2017-09-24:
I think it's max that is interfering with flash's SPI transaction when it is not activated by the corresponding driver.

I concluded this as flash is detected and usable when max is initialized and working.
`spiPreInitCs` was introduced to prevent this situation, but it is apparently not working.

mikeller, 2017-09-24:
Ah I see. Sounds reasonable. Shouldn't just pulling CS high stop the MAX from doing anything on the bus? Or is MAX another one of those 'borked SPI' chips that won't behave?

jflyper, 2017-09-24:
> > Shouldn't just pulling CS high stop the MAX from doing anything on the bus? 

To date, MAX is not considered as borked chip. However, interestingly, while majority of new designs use AT, Kakute F4 is one of the design that continue to use the original MAX. I'm wondering if MAX/AT difference (again) makes this happen. It's not definite yet; it may be some board design related.

First, we can find 'max interfering with flash' hypothesis holds.

`spiPreInitCs` only pulls up weak through `IOCFG_IPU`.
I will make a temporary PoC firmware that pulls this hard by `IOLo`.

jflyper, 2017-09-24:
@Haasje @mikeller 
This is a firmware that pulls up CS for MAX.

[betaflight_3.2.0_KAKUTEF4-maxhardpull.hex.zip](https://github.com/betaflight/betaflight/files/1327323/betaflight_3.2.0_KAKUTEF4-maxhardpull.hex.zip)

hydra, 2017-09-24:
the rtc6705 is also another device that must have it's CS pin high BEFORE the rest of the SPI pins are initialised.

in my experience order should be:
1. set all CS pins so that all devices are DISABLED
2. configure SPI clock pin BIT so that when the clock PIN is initialized the clock signal is correct
3. configure remaining SPI pins.

Haasje, 2017-09-24:
@jflyper I will try that firmware tonight and let you know

jflyper, 2017-09-24:
@hydra Yes. RTC6705 is a very well known borked chip. However, we believe that the current code does pulls up all CS pins with `IOCFG_IPU`, during `init` via `spiPreInit` and `spiPreInitCs`, before SPI initialization.

The hex above change the pull-up method to hard pull-up.
```
void spiPreInitCsHard(ioTag_t iotag)
{
    IO_t io = IOGetByTag(iotag);
    if (io) {
        IOInit(io, OWNER_SPI_PREINIT, 0);
        IOConfigGPIO(io, IOCFG_OUT_PP);
        IOHi(io);
    }
}
```
instead of
```
void spiPreInitCs(ioTag_t iotag)
{
    IO_t io = IOGetByTag(iotag);
    if (io) {
        IOInit(io, OWNER_SPI_PREINIT, 0);
        IOConfigGPIO(io, IOCFG_IPU);
    }
}
```

https://github.com/jflyper/cleanflight/tree/bfdev-kakutef4-max-flash-interference

mikeller, 2017-09-24:
@jflyper: Tried the firmware from https://github.com/betaflight/betaflight/issues/4223#issuecomment-331708369. The dataflash works even when powering up with OSD disabled. Looks like we got a winner. Do you think it is the MCU's internal pull ups being below spec, or the MAX chip pulling too hard?

Haasje, 2017-09-24:
@jflyper I also tried the firmware and the dataflash keeps working with OSD on and off even after power cycles so this seems to fix it indeed.

jflyper, 2017-09-25:
@mikeller Okay, we now know how to solve it, but we don't know what caused it. It may even be a board design...

mikeller, 2017-09-25:
Board design as in pull down on OSD CS? @jamming is normally following in here, and he should be able to answer that (without having to resort to tracing efforts).

jflyper, 2017-09-25:
Okay, we will wait for him. This is actually not a big deal, because there is a workaround 😉

jamming, 2017-09-25:
Onboard flash and osd are on the same bus SPI3. They have different chip selects.  Chip selects are directly tied to MCU without external pullup/pulldown resistors.
MAX7456_CS <----->PB14
FLASH_CS <----->PB3

jflyper, 2017-09-25:
@jamming Thanks for the info.

So my theory is "MAX7456 CS pin is sinking more than what `IOCFG_IPU` can supply.

jflyper, 2017-09-25:
It may be the case that

"_**An SPI slave device must be clocked for a while with CS being held high to get properly initialized**_"

IIRC, there are couples of devices like this...

mikeller, 2017-09-25:
> "An SPI slave device must be clocked for a while with CS being held high to get properly initialized"

How would the difference between using internal pull up and high output cause a difference in behaviour? Rise time?

jflyper, 2017-09-25:
> > "An SPI slave device must be clocked for a while with CS being held high to get properly initialized"
>
> How would the difference between using internal pull up and high output cause a difference in behaviour? Rise time?

Doh! Good question. But I made the PoC firmware anyway 😢 

[betaflight_3.2.0_KAKUTEF4-spi_dummy_clocking.hex.zip](https://github.com/betaflight/betaflight/files/1328325/betaflight_3.2.0_KAKUTEF4-spi_dummy_clocking.hex.zip)

jamming, 2017-09-25:
When

jamming, 2017-09-25:
I test it with 3.2.0 master, when OSDfeature is disabled, PB14 is pull low.

May be PB14 is redefined in some where.

In "Bus_spi_pinconfig.c",there are codes:

#ifndef SPI2_SCK_PIN
#define SPI2_SCK_PIN    PB13
#define SPI2_MISO_PIN   PB14
#define SPI2_MOSI_PIN   PB15
#endif

jflyper, 2017-09-25:
@jamming 

> May be PB14 is redefined in some where.
> In "bus_spi_pinconfig.c",there are codes:
> #define SPI2_MISO_PIN   PB14

This is only valid when `USE_SPI_DEVICE_2` is defined, which is not a case for KAKUTEF4.

> I test it with 3.2.0 master, when OSDfeature is disabled, PB14 is pull low.

We have took elsewhere ...

mikeller, 2017-09-25:
> Doh! Good question. But I made the PoC firmware anyway 
> ￼betaflight_3.2.0_KAKUTEF4-spi_dummy_clocking.hex.zip

Well, eliminating things that are not the cause will bring us closer to a solution as well.
Which is to say that the above firmware shows the behaviour that the dataflash is **not** recognised after a cold boot with OSD disabled.

(As a working hypothesis, how about we assume it is indeed a case of the pull ups not being strong enough to give the MAX7456 a high. What would be the downside of setting the CS pins to Hi instead of using the pull ups?)

jflyper, 2017-09-25:
> (As a working hypothesis, how about we assume it is indeed a case of the pull ups not being strong enough to give the MAX7456 a high. What would be the downside of setting the CS pins to Hi instead of using the pull ups?)

It seems that the hypothesis doesn't generally hold.

I did a test with NUK-F405RG + MAX7456 breakout, flashed with KAKUTEF4 target (acc/gyro substituted with fake acc/gyro to shorten the boot time) from current master. The CS line for the MAX7456 is straight out of PB14.

OSD enabled
[image]

OSD disabled (MAX7456 Vdd=5V)
[image]

OSD disabled (MAX7456 Vdd = 3V3)
[image]

PB14 is pulled high even when OSD is disabled.

mikeller, 2017-09-25:
That's with pull up, or with output high for CS? How do the timings compare between these 2?

jflyper, 2017-09-25:
These are without an external pullup, just CS straight out of PB14. Time scale is identical between two. The little spike on SCK and MOSI at around 1sec is probing for flash (which doesn't exist), I think.

mikeller, 2017-09-25:
So CS uses internal pull up, or IOHi? (I'm wondering if the two result in a difference in timing.)

jflyper, 2017-09-25:
These all use internal (`IOCFG_IPU`).

jamming, 2017-09-25:
There is a voltage level shift TXB0104UQFN between F405 and MAX7456. SPI bus level on F405 side is 3.3v, and it is 5V on MAX7456 side.

mikeller, 2017-09-25:
@jflyper: Can you scope the same setup, except with the IOHi modification from https://github.com/betaflight/betaflight/issues/4223#issuecomment-331708369?

jflyper, 2017-09-25:
@jamming 
Isn't TXB0104 current driven?

jflyper, 2017-09-25:
Okay, I think the existence of the TXB0104 explains why `IOHi` works and `IOCFG_IPU` doesn't.

Like most level shifters do, TXB0104 requires current flowing to keep the driver activated.

>http://www.ti.com/lit/ds/symlink/txb0104.pdf
>p.13
> 8.1 Overview
> This device can only translate push-pull CMOS logic outputs.
>
>p.14 
> 8.3.2 Input Driver Requirements
>Typical IIN vs VIN characteristics of the TXB0104 are shown in Figure 6. For proper operation, the device driving the data I/Os of the TXB0104 must have drive strength of at least ±2 mA.

These requirements can only be met with active PP output, not with `IOCFG_IPU`.

I suggest that we treat the `IOCFG_OUT_PP` with `IOHi` to pre-initialize the CS for MAX7456 only.

jflyper, 2017-09-25:
I thought the 5V fed original MAX could interface with 3V3 devices without a problem... well, it must have been some design decision.

mikeller, 2017-09-25:
> I suggest that for 3.2, we treat MAX7456 special, and pre-initialize CS with IOCFG_OUT_PP with IOHi instead of IOCFG_IPU.

Sounds like the reasonable approach. For 3.3 we probably want a `csIsPp` parameter for the SPI device struct.

jflyper, 2017-09-25:
This is the hex from the PR, if ppl want to test it.

[betaflight_3.2.0_KAKUTEF4-PR4234-00.hex.zip](https://github.com/betaflight/betaflight/files/1330059/betaflight_3.2.0_KAKUTEF4-PR4234-00.hex.zip)

jamming, 2017-09-26:
See the following description.

Haasje, 2017-09-26:
I just wanna say that I'm truly impressed with the way you guys picked this up and found the solution. I've been in various beta testing programs and github projects, but the speed and level of profession with which you picked this up, kuddo's to you!!

jflyper, 2017-09-26:
Closed via #4234

closed 2017-09-26 by jflyper