betaflight/betaflight #4226: Acc filtering is broken.

hydra, 2017-09-24:
After examining the startup code I found acc filtering to be broken.

`activateConfig()` calls `setAccelerationFilter.`
` acc.accSamplingInterval` is used to set the filter.
however, `acc.samplingInterval` will not have been configured yet.
`acc.samplingInterval` is set by `accInit()`
`accInit` is called by `sensorsAudtoDetect()`

it looks like the filter will never be initialised.

Wtcheesecakes, 2017-09-25:
would this explain when updating from rc3 or rc4 (cant remember) to rc5 and am now getting horrible oscillations, would this be the issue?

isilcala, 2017-09-26:
I have a 1s quad which gets horrible oscillations after flashing a very recent nightly build. Not sure if this is the cause.

modellbobby, 2017-10-01:
Hi, it is still present in rc5. Any quick fix here available?

hydra, 2017-10-02:
this is fixed by https://github.com/betaflight/betaflight/pull/4218

closed 2017-10-02 by hydra

modellbobby, 2017-10-02:
awesome . Very keen to test it now

VoicOfReason, 2017-10-02:
Does RC6 have this fix included?

VoicOfReason, 2017-10-02:
I just upgraded to RC6, but it isn't fixed. How do I incorporate this fix in my RC6 build?

modellbobby, 2017-10-20:
Dear Des, can somebody please fix this? It is really screwing the angle mode

VoicOfReason, 2017-10-20:
Agreed.

hydra, 2017-10-22:
this particular issue is closed, if there is a still an issue with angle/horizion please create a new issue.  This issue was for a specific technical issue which was resolved with #4218.