betaflight/betaflight #4236: Use Dshot commands to switch 3d On/OFF and change motor spin directions

brycedjohnson, 2017-09-25:
Messing around with #4227 has made me want to have a more userfriendy way to set up 3D mode or just reverse the direction of one motor using betaflight using dshot commands (for setting up a quad).  

I know we can currently do that functionality with the CLI and dshotprog commands.

Betaflight could keep track on what motors you want normal and reverse for setup and send it down to the quad.  Also keep track of bidirectional or bidirectional reverse for 3d mode.  If you kept track of both of those you could switch between 3D and normal flying mode on a switch.

mikeller, 2017-09-25:
Betaflight doesn't even have to keep track of the settings. At least for KISS and BLHeli32, it can just load the settings from the ESCs, and then store the updated settings on the ESC again. The current CLI implementation is just a proof of concept, with the idea to later on add an `MSP_DSHOT_COMMAND` MSP command.

dk7xe, 2017-12-11:
is there something available to be tested on the SPRACINGF4 target already? i don't care if it is CLI or  MSP_DSHOT_COMMAND but i have just replaced the KISS24A on my Octo-Monster by KISS32A and have not set the jumpers before putting the ESC into the heat shrink tube. stupid me. crossing two motor cabels does not look that nice ;)
[image]

brycedjohnson, 2017-12-11:
Should be able to use dshotprog in the cli to set the direction and save
settings in any recent betaflight.  dshotprog motornum dshotcommand

dk7xe, 2017-12-13:
@brycedjohnson may you please give me one example on how to specify the spin direction reversal for e.g. motor 5
I tried "dshotprog 5 8" because i have found somewhere that command 8 is spin direction reversal.
But i ended up with beeping ESC's. Thank you!

mikeller, 2017-12-13:
@dk7xe:
- ESC numbers are 0 based in `dshotprog`, i.e. motor 5 needs `dshotprog 4 ...`;
- the list of commands is here: https://github.com/betaflight/betaflight/blob/f9feb03a93d73b8aec325e9b8b668062dbe11bf7/src/main/drivers/pwm_output.h#L40

i.e. `8` is correct, but you need a `12` / `DSHOT_CMD_SAVE_SETTINGS` to save your settings afterwards.

dk7xe, 2017-12-13:
Thanx @mikeller ! 
The beeping ESC's after issuing `dshotprog 5 8` made me worry yesterday. Had a discussion with @ronlix afterwards and the beeps are indicating corrupt DSHOT signal. Maybe the DSHOT signal is getting corrupt by an DMA error after issuing the command.
Today motors were running ok, no beeps at all, except that i still have 4 motors turning into wrong direction. Your post reached me while i was already on the way to the office therefore it will take another day until i can sort that out.
I run BF3.2.0 with DSHOT600 on the SPRF4 btw.

dk7xe, 2017-12-14:
Because of https://github.com/betaflight/betaflight/issues/4748 there is a delay in my testing. Ist's not directly related but i like to get that sorted out.  Most probably i will revert back to BF3.2.0

dk7xe, 2017-12-15:
works! issued `dshotprog 4 8` followed by `dshotprog 4 12` to revert motor 5. Did this for motor 6,7&8 as well.
ESC's were updated to FW 118e before and FC is now on BF3.2.3
Will go flying within the next hour ;)
Thank you!

closed 2018-06-23 by brycedjohnson