betaflight/betaflight #4237: OMNIBUS SD F4 V5

mikeymac22, 2017-09-25:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://rcgroups.com/forums/showthread.php?t=2464844), [Slack](https://slack.betaflight.tech) or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1.) Describe your problem;
2.) Include ways to reproduce the problem;
3.) Provide information about your flightcontroller and other components including how they are connected/wired;
4.) Add the used configuration and firmware version;
5.) Create a `diff` and post it here in a code block. Put \`\`\` (three backticks) at the start and end of the `diff` block; and
6.) Remove this Text :).
Flashing RC5 seems to flash properly however it closes the serial port but it wont allow thw FC to open back up under Betaflight GUI. Not sure what the issue is but its not working properly.

closed 2017-09-25 by DieHertz

mikeymac22, 2017-09-25:
Flashing RC5 seems to flash properly however it closes the serial port but it wont allow thw FC to open back up under Betaflight GUI. Not sure what the issue is but its not working properly.

mikeymac22, 2017-09-25:
why is this closed. Just wondering.

DieHertz, 2017-09-25:
This issue looks like a support request, a genuine Omnibus f4 SD has no issues booting and connecting on RC5.
Please redirect your questions towards a corresponding RCG discussion thread.

mikeymac22, 2017-09-25:
ok. close the other one i opened please. I will check into it at other group, Thanks anyway and sorry my friend.

richard-scott, 2017-09-26:
@mikeymac22 Once flashing ALWAYS disconnect and reconnect USB.