betaflight/betaflight #4239: X4R Sbus issues in Betaflight

katana144, 2017-09-25:
Hello, 
     I am having trouble getting my x4r sbus receiver to respond to commands from my frsky qx7. Initially i was trying to set up switches with my new build and am using a Hobbywing f4 fc. I bound the receiver and transmitter and then went into betaflight (configurator 3.2.1). I flashed 3.2.0 onto the flight controller and then set up the receiver by following the instructions given on by hobbywing for the flight controller. I selected serial rx under uart1, set receiver mode to "serial based" and then set provider to "sbus." I then went to to the receiver tab in betaflight but none of my stick inputs registered. 
     Thinking i had a bad receiver i swapped out the x4r on my new build for the one on my vortex 250 and still no response from my stick commands in betaflight.
     I then plugged in my working vortex 250 with the same reciever paired to the same transmitter (which has been working for months) to see if i set up anything differently and suddenly betaflight switched my "serial receiver provider" to "spektrum" and I haven't been able to get it to stick to "sbus" since. I tried entering in the command, "set serialrx_provider = SBUS" into CLI and then saved and backed out and still it reset it to "spektrum." I have double and triple checked my wiring on my new build and it is all correct according to the pad layouts given by the company. Prior to plugging my vortex 250 into betaflight it had been flying for months without a problem. 
     Is there a step I am missing in terms of setting up an sbus receiver?

mikeller, 2017-09-25:
The symptoms sound like a corrupted configuration. Back up your config in CLI, reset to defaults, restore your config, and test again.

katana144, 2017-09-25:
How do I backup and reset to default?

mikeller, 2017-09-25:
Go into CLI, `diff all`. 'Save to File'. `defaults`. After reboot, copy / paste the saved file (minust the line that reads `defaults`) into CLI, hit enter.

katana144, 2017-09-25:
I went into settings, clicked " reset setting" and then re-entered the configurations via CLI and while the receive tab will read my initial throttle position (as it had before), it will not register any change to any stick movements.

jflyper, 2017-09-25:
What is the version of the firmware? Type CLI "version" to find out.

Arakon, 2017-09-26:
Are you sure UART1 is the SBUS receiver port? F4 CPUs don't do serial inversion on-chip, so you need to connect the receiver to the port that has an inverter, or it won't work.

mikeller, 2017-09-26:
Ha, well spotted @Arakon. From the looks of it, this is an OMNIBUSF4SD  clone, so the SBus RX will have to go onto UART6.

katana144, 2017-09-26:
That worked. I got it to respond.
Thank you very much.

closed 2017-09-28 by jflyper