betaflight/betaflight #4242: Blackbox Log Number for internal flash

alexeystn, 2017-09-27:
Hi!
I have DYS Omnibus F4 Pro with onboard flash memory and Betaflight 3.2 RC5 firmware.
I can see Blackbox usage and Blackbox log number (BB LOG NUM) after each disarm on the OSD stats screen.
Usage is displayed correctly in %, while Log Number is always 0.
In the source code I've found that _blackboxGetLogNumber_ function always return zero if onboard flash is used. 
Is it possible to display real number of log entry on the stats screen? It would be easier to match the DVR recordings with corresponding logs in Blackbox Explorer.
Thanks.

alexeystn, 2017-09-27:
Here are the pictures of how it looks now.
Would be cool if the number on OSD is the same as in BB Explorer for every arm cycle.
[image]
[image]

mikeller, 2017-09-27:
Totally agreed. This is not supported yet, but definitely worth looking into for 3.3. It will need a bit of work though, since we are currently not keeping track of the number of logs written to the internal dataflash.

bkeinert, 2017-09-30:
What's the best way to implement this? Had a rough look at the BB code, it seems the start of the free flash is found by a binary search. Maybe counting "End of flight" events? But that would be O(n) - might be still fast enough.

jflyper, 2017-09-30:
Plan 1

An idea would be to insert a dedicated page that keeps an ordinal for the following log chunk.

Page 0: <log ordinal #x>
Page 1: <page 0 of log #x>
Page 2: <page 1 of log #x>
Page n: <page n - 1 of log #x>
Page n + 1:  <log ordinal #x + 1>
Page n + 2: <page 0 of log #x + 1>
Page n + 3: <page 1 of log #x + 1>
Page n + m: <page n + m - 2 of log #x + 1>
Page n + m + 1: <log ordinal #x + 2>
Page n + m + 2: First available page

- When flashfs locates the first available page, it takes a look at the previous page, and detect the current log ordinal, pass it to OSD and so on.
- When flashfs closes the log chunk, it will append a new ordinal page.
- If the first available page is the page zero, then log ordinal is set to zero, and an ordinal page is written. first available page advances by one.
- if the first available page is not page zero, and the previous page is not an ordinal page, then this is a trouble. Have to do some expensive linear search backwards until the most recent ordinal page is found.

bkeinert, 2017-10-01:
Maybe write the number of the current log at the beginning and the end of the currently active log?
Doing so would at least allow you to recover a valid log number even if logging was interrupted (e.g. crash and flown away lipo etc.).

mikeller, 2017-10-05:
As a much more simplistic approach, why not store the current log number in a fixed location (beginning / end), and increase it by 1 every time a new log is started?

jflyper, 2017-10-05:
Fixed location on the flash?

mikeller, 2017-10-06:
Yeap. Or even in the EEPROM.

shkatulo, 2019-06-12:
It should be great feature, especially, when you have to analyze a lot of flying data with videos.
Nobody solved this yet?

mikeller, 2019-06-12:
Different suggestion: Now that we have nice persisted and accessible permanent flight number logging (#7927), we could / should probably add these stats to the blackbox header, and use the overall flight number in the file name (assuming that statistics are enabled).

Quickred, 2019-10-31:
I would also like to have this feature, showing the log number for blackbox for onboard blackbox. For me, as a beginner in blackbox analyzing, I found it confusing, to see always a "0" in the osd. If it is not possible, than it is better to remove the number and show only the indicator for recording

mikeller, 2019-11-02:
@Quickred: That's much easier to fix: #9141

mituritsyn, 2022-11-14:
@alexeystn still relevant?)

closed 2026-09-26 by blckmn