betaflight/betaflight #4249: Access RunCam OSD Menu via Camera Control

TempleClause, 2017-09-28:
Camera Control works great for me and is an awesome feature!

RunCam also has some other menus that one can access via the OSD push button cable. 
For example, the new RunCam Micro Swift 2 has presets for indoor, cloudy etc conditions. 
You can change those settings by long pressing the UP button. 

Is this something that is being worked on? I guess it should be quite easy to implement since it only needs a new stick command which then presses UP instead of the SET key.

TempleClause, 2017-09-28:
Case closed :-) 

Just found in the source code that if you have throttle low, yaw center, pitch high and roll center it will access the menu. To navigate afterwards the throttle needs to get back to center.

closed 2017-09-28 by TempleClause

DieHertz, 2017-09-28:
There will be something better this weekend, allowing you to hold sticks in position to get the same behavior as long press

Mmarsian, 2017-11-03:
Sorry to pull this old one up. Has this "long press" feature found its way into current Betaflight?
I played around with the sticks but never could access the presets of the Runcam Micro Swift 2.

bhameline, 2017-11-13:
I also want to bring this issue up but on the exiting procedure. I can get into the menu with the throttle down, yaw center, roll center and pitch up. I can move through the RunCam OSD menu, but the "enter" function no longer works. Roll and pitch function normally. Ideas? I am specifically concerned with pressing enter on the "exit" button.

DieHertz, 2017-11-13:
No longer works? It never worked in first place then, add a capacitor.