betaflight/betaflight #4257: ESC init happens twice on power-up

Arakon, 2017-09-30:
Using a current master build, when using BLHeli 16.63 or 16.67 with standard startup tones, the startup sounds play twice. When using a custom tone in 16.67 (i.e. the star wars startup melody), it doesn't happen.
Using DShot 600. Thought it was just a fluke on my system, but someone else just reported the same in the BF thread on rcgroups.

dannyteets, 2017-10-01:
Having same issue. DShot 600 in RC6.

dannyteets, 2017-10-01:
Holybro Kakute F4 FC.

richard-scott, 2017-10-01:
I get the same on an SP Racing F3 Evo. three tones, three tones then dshot tones.

0crap, 2017-10-02:
Same here.
Naze32 - FVT LittleBee 20A - OneShot125

devbloggs, 2017-10-02:
same issue here, startup tones happen twice when powered with LiPo, but only once when connected to betaflight with LiPo, didnt happen on RC5

Diatone DLink F3 V3
DYS XS20A DSHOT 600 (tried 150 and 300)

0crap, 2017-10-02:
I flashed a bunch of builds on my Naze32 to find out where it went wrong.
Conclusion, build #224 is the last one without this issue.
Build #225 "Changes: Improved initialisation order. Stopped calling pidInit before gyro" is the first build which gives me the above issues.

Hope this makes it a bit easier for the dev's to find the issue.

richard-scott, 2017-10-02:
I've noticed that when you go into CLI, and click out of CLI (or type save) the board reboots but only beeps the correct number of times.

VoicOfReason, 2017-10-04:
I’m having double initialize as well: omnibus f7v2, dshot 600 16.63.

snowblind64, 2017-10-05:
Same here. Omnibus F4 Pro with BF 3.2 RC6 with BlHeli_S 16.63 and DShot 600. Doesn't happen on my F3 boards or before updating to RC6. Doesn't seem to have any impact on flight performance or cause any other issues. Just a bit disconcerting. At first I thought there was a loose connection somewhere.

universe360, 2017-10-05:
Same issue here on DYS F4 Omnibus BLHeli_S 16.63, DShot 600 with double initialization...  just resoldered cold solder joints an meter checked signal pads esc, bec.. then I remembered it was fine before updating.. Glad to see its a software related and not faulty hardware.

kolossboss, 2017-10-06:
Same issue with Omnibus F4 v2 OneShot42

Detegr, 2017-10-07:
Same issue here, Omnibus AIO F3, OneShot125.

AndersHoglund, 2017-10-07:
There is one short narrow pulse on BFF3 before the real ESC signaling starts running. See the attached overview and closeup. Hard to see on the overview, but it is there under the orange trig arrow.
On DYSF4PRO, there is a much much wider pulse and a gap before signaling starts.
[image]
[image]
[image]

frdstang93, 2017-10-08:
Same here, CL Racing F4 AIO, Dshot 600

snowblind64, 2017-10-08:
Seems to be specific to blheli 16.63. I flashed 16.67 and no longer have double init.

ESC beeper also working fine. Just a CLI command I missed. On RC6 forward the CLI command "set beeper_dshot_beacon_tone = 1-5" is needed.

VoicOfReason, 2017-10-08:
Thanks!! Saved me some trouble. I've been having those same issues.

Detegr, 2017-10-08:
> Seems to be specific to blheli 16.63. I flashed 16.67 and no longer have double init.
Flashing 16.67RC with BLHeli Configurator fixed the issue for me as well, thanks!

AndersHoglund, 2017-10-08:
Still, we should not kick the ESC with spurious pulses during startup, even if the sensitivity to it is different on different versions of  BlHeli. 
Also a version with startup music will hide this dual init.

Rreaper86, 2017-10-08:
AndersHoglund 

I Agree.
There is a problem and i think its not the esc's.
if that was the case then it hould not have effected it when upgrading the BF firmware.

There is a problem somewhere and it needs to be fixed before i start flying with it.
going back to RC5 for now.

0crap, 2017-10-08:
Try going between build 224 en build 225.
For me build 224 is the last one without this issue.
This has nothing to do with BLHeli.

Unfortunately this bug is now in the 3.2.0 stable.
Guess no fix for this for Naze32 boards. :(

0crap, 2017-10-11:
Tested build 249 and this fix works on my Naze32.
Thx all involved!

richard-scott, 2017-10-15:
I can confirm that this is fixed for me too.

closed 2017-10-15 by mikeller