betaflight/betaflight #4258: BF3.2 RC6 Latest Built - DShot Beeper Command (Aux Channel) Not working

TehllamaFPV, 2017-09-30:
I just started moving my quads over to the 3.2 Release Candidate 5 over the last few hours.  All but one of them works perfectly (on identical setups), with every new feature working.

The last one came up with RC6 as the download in the configurator - it does everything except for the Aux channel Beeper functionality.

Arakon, 2017-10-01:
It's working fine. You just have to enable it via CLI.
set beeper_dshot_beacon_tone = 1-5 (different tones)
Tested it just now too.

TehllamaFPV, 2017-10-01:
I just went back to RC5 - which is working for me.
Awesome - I feel dumb...  although a GUI button for selecting those tones from a drop-down would be pretty awesome; although a low priority feature request at best.
Thanks!

dingrgh, 2017-10-01:
Let's close this, shall we?

AdrY89, 2017-10-01:
Hi
I have a KAKUTE F4
The rc6 the beeper channel work only with esc if set 
set beeper_dshot_beacon_tone = 1-5
the standard beeper do not work

closed 2017-10-05 by mikeller